zachjs/sv2v #6: Macro defined in package not found

dpetrisko, 2019-08-02:
foo_pkg.vh
```
package foo_pkg;
  `define width_calc(a) (a+3)
endpackage
```
foo.v
```
module foo
import foo_pkg::*
 #(parameter width = `width_calc(2))
  (input [width-1:0] i
   , output [width-1:0] o
   );

 assign o = i + 1'b1;

endmodule
```
Command:
```
$ sv2v foo_pkg.vh foo.v
```
Result:
```
sv2v: Lexical Error: Undefined macro: width_calc, at ./foo.v:4:33
CallStack (from HasCallStack):
  error, called at src/Language/SystemVerilog/Parser/Lex.x:346:21 in main:Language.SystemVerilog.Parser.Lex
```

However, this works:
bar.v:
```
`define width_calc(a) (a+3)

module bar
 #(parameter width = `width_calc(2))
  (input [width-1:0] i
   , output [width-1:0] o
   );

 assign o = i + 1'b1;

endmodule
```
Command:
```
sv2v bar.v
```

Result:
```
module bar (
	i,
	o
);
	parameter width = (2 + 3);
	input [(width - 1):0] i;
	output [(width - 1):0] o;
	assign o = (i + 1'b1);
endmodule
```

zachjs, 2019-08-03:
Thank you for your detailed report! Your example highlights some interesting aspects of SystemVerilog's specification with regard to how preprocessor directives (like ``` `define``` and ``` `ifdef```) are handled.

Firstly, these directives have no regard for traditional scoping constructs, including `module`, `package`, `function`, `task`, etc. Once something is ``` `define```d, it is available for the remainder of the "compilation unit" (a term I'll discuss later). For example, the following is valid:
```systemverilog
module foo;
    `define X 1
endmodule
module bar;
    parameter P = `X;
endmodule
```
Of course, if `X` was instead a parameter, that would not be valid. As a further example, the following is also valid:
```systemverilog
module
`define X 1
bar;
    parameter P = `X;
endmodule
```
So SystemVerilog really has no concept of a macro being defined "within a package." SystemVerilog's behavior in this regard matches that of C/C++. So, under the specification, the `package` in your example does not have the intended affect.

Another issue worth noting is that `import` statements are not allowed in the middle of the module "header." sv2v doesn't complain about this syntax error because preprocessing necessarily occurs before parsing, and so it encounters the undefined macro before it attempts to parse to file. One way to get around this limitation would be to do something like:
```systemverilog
module foo;
    import foo_pkg::*;
    parameter blah = something_imported;
    input [blah-1:0] i;
    output [blah-1:0] o;
    ...
endmodule
```

_Setting those two issues aside,_ I think you've highlighted something in sv2v that might need to be changed. Someone might reasonably expect the following to work:
```systemverilog
/*** file: foo.vh ***/
`define X 1

/*** file: foo.v ***/
module bar;
    parameter P = `X;
endmodule

/*** convert using: sv2v foo.vh foo.v ***/
```
But, as of now, sv2v will still complain that the macro is undefined.

The specification states that a "compilation unit" is "a collection of one or more SystemVerilog source files compiled together." It goes on to explain that tools should provided a mode in which the compilation unit is the set of _all_ files passed on the command line, and a mode in which each file is effectively compiled separately. In some sense, sv2v blends these two modes.

For example, an instantiation of an interface defined in a separate file will indeed be successfully converted. When trying to figure out what the definition of an interface is, sv2v first looks for definitions of the interface within that same file (and the files that it includes), and if it is not found, it looks across all the files it is given. In essence, definitions within the file take precedence over those found elsewhere. Given that hardware designers often define dependent modules, interfaces, etc., in different files, I think this behavior is pretty reasonable.

However, macros throw a wrench into this model. If macros were processed this way, a user might discover that a macro defined in one file is unexpectedly defined in another. For example:
```systemverilog
/*** file: a.sv ***/
`define DO_SOMETHING_SPECIAL
...
`ifdef DO_SOMETHING_SPECIAL
    assign foo = 0;
`else
    assign foo = 1;
`endif
...

/*** file: b.sv ***/
...
`ifdef DO_SOMETHING_SPECIAL
    assign bar = 0;
`else
    assign bar = 1;
`endif
...

/*** convert using: sv2v a.sv b.sv ***/
```
I think most users would expect the first assign to be used in `a.sv`, and the second assign to be used in `b.sv`, as that macro isn't really defined in the context of that file. But if the first file were instead a header, users would probably expect the macro to be defined in both files.

I didn't anticipate that users would explicitly pass sv2v header files. Rather, I believed source files would explicitly ``` `include``` the headers they needed, potentially specifying include search paths for sv2v using the `-i`/`--incdir` flag. I now believe that explicitly passed header files should be handled specially, but I'm not sure what the appropriate behavior would be.

I'm assuming that `.vh` and `.svh` unambiguously denote header files.

Would it be reasonable for sv2v to implicitly ``` `include``` explicitly provided header files at the beginning of every non-header file it is given? This could lead to the header being unintentionally included multiple times in the same file. I'm not sure if include guards are as commonly used in SystemVerilog as in C and C++. Should sv2v print some kind of warning to make users aware of this implicit behavior? Should there be a flag which disables this behavior? If a user ran `sv2v a.svh b.sv c.svh d.sv`, should the headers be implicitly included before all of the non-header files, or only specific ones?

dpetrisko, 2019-08-03:
Thanks for the quick and thorough response. In SystemVerilog 2011, it is legal to have package import statements inside of modules, as shown. 

http://www.ece.uah.edu/~gaede/cpe526/2012%20System%20Verilog%20Language%20Reference%20Manual.pdf#page787

This has an example of that from the LRM (it also works in all Synopsys tools, haven’t tested with Cadence). More importantly, it actually functions differently than having the imports outside the module. 

Having the package import outside the module indeed imports at $unit scope, whereas inside the module the package is l scoped to within the module. There are subtle differences if you do an ‘include instead of an import (in local scope) where some comparison operations will inexplicably fail due to scoping. 

As for tick defines being globally scoped, that’s of course true and my wording was careless. My actual use case is having localparameters defined in the package file, but I thought this was a simpler counterexample. 

Generally, (this is tool behavior, not LRM as far as I know) passing packages first in the file list makes it so the tool processes them first. In this case, I would expect the package to be processed and the globally defined macro to be set before moving on to foo.v

zachjs, 2019-08-04:
sv2v does indeed have basic support for packages, whether imported at the compilation unit scope or otherwise. However, I was unaware that imports were allowed within module _headers_. I have added support for this on master (with some test coverage). Thank you for bringing this to my attention!

I've also now added some explicit coverage for package resolution across files based on the example you provided. I used a function rather than a macro. You can find that new test [here](https://github.com/zachjs/sv2v/tree/8119cdd6b11703a6ef3998ff3888d22c1f448bd7/test/resolve).

I will look into some different options for handling header files in a different way during preprocessor.

dpetrisko, 2019-08-04:
Awesome, thanks!!

dpetrisko, 2019-08-05:
```
Would it be reasonable for sv2v to implicitly `include explicitly provided header files at the beginning of every non-header file it is given? This could lead to the header being unintentionally included multiple times in the same file. I'm not sure if include guards are as commonly used in SystemVerilog as in C and C++. Should sv2v print some kind of warning to make users aware of this implicit behavior? Should there be a flag which disables this behavior? If a user ran sv2v a.svh b.sv c.svh d.sv, should the headers be implicitly included before all of the non-header files, or only specific ones?
```

From what I've seen, include guards are fairly common in SV code for `define type headers, so I don't think that's a problem.  Possibly there could be issues with treating packages as headers with multiply defining packages.  Your DO_SOMETHING_SPECIAL example is a good one for unintuitive behavior, but I don't think it's all too common in practice.  More often I see the style of a bunch of global defines in one file which is included by all the others or defining per file.

Would it help to have our packages as *.v rather than *.vh?

dpetrisko, 2019-08-05:
And to clarify the way we're using (abusing :)) SystemVerilog for our project, we're essentially using packages to group a series of `include files along with any global parameters.  Then we just import the package and all macros should be defined along with constant and macro-generated parameters.

https://github.com/black-parrot/pre-alpha-release/blob/master/bp_top/src/v/bp_addr_map.v
https://github.com/black-parrot/pre-alpha-release/blob/master/bp_common/src/include/bp_common_aviary_pkg.vh
https://github.com/black-parrot/pre-alpha-release/blob/master/bp_common/src/include/bp_common_aviary_defines.vh

zachjs, 2019-08-06:
> Would it help to have our packages as *.v rather than *.vh?

No, sv2v wouldn't react to this change in its current form. Further, I have been trying to avoid requiring users to change their code to work with sv2v, unless there is genuine disagreement with the specification.

You will now find a new option called `--oneunit` on master. This makes macros found in earlier files available later on. I have it behind this flag to preserve the existing behavior by default.

Please let me know if this works for you!

dpetrisko, 2019-08-06:
> No, sv2v wouldn't react to this change in its current form. Further, I have been trying to avoid requiring users to change their code to work with sv2v, unless there is genuine disagreement with the specification.

Very much appreciated :)

dpetrisko, 2019-08-06:
Appears to have worked!

closed 2019-08-06 by zachjs