zachjs/sv2v #7: Support for localparams in parameter list

dpetrisko, 2019-08-06:
```
module foo
 #(parameter baz = 2
   , localparam in_width_p = baz + 2
   , localparam out_width_p = baz + 3
   )
  (input [in_width_p-1:0] a
   , output [out_width_p-1:0] b
   );

assign b = a+1;

endmodule
```

I think this feature came in SV-2012
http://www.ece.uah.edu/~gaede/cpe526/2012%20System%20Verilog%20Language%20Reference%20Manual.pdf
Section 06.20.04 
>Local parameters may be declared in a module’s parameter_port_list.

Possibly a solution is to convert localparams in parameter lists to vanilla parameters?

zachjs, 2019-08-06:
Thank you for another great report! Support has been added.

closed 2019-08-06 by zachjs

taylor-bsg, 2019-08-07:
Loving these fixes — sv2v is going to be awesome!

commit e82537fa (2019-08-06): add support for localparams in module parameter list (resolves #7)