zachjs/sv2v #8: Support for array of struct parameters

dpetrisko, 2019-08-07:
foo.v
```
typedef struct packed
{
  logic [2:0] width;
} baz_s;

parameter baz_s [1:0] bar = '{2, 1};

module foo
 #(parameter width_p = bar[0])
  (input [width_p-1:0] x
   , output [width_p-1:0] y
   );

   assign y = x + 1'b1;

endmodule
```

Command:
```
sv2v --oneunit foo.v
```

Result:
```
sv2v: Parse error: unexpected token '[' (Sym_brack_l) at ./foo.v:8:17.
CallStack (from HasCallStack):
  error, called at .stack-work/dist/x86_64-linux/Cabal-2.4.0.1/build/sv2v/sv2v-tmp/Language/SystemVerilog/Parser/Parse.hs:6124:22 in main:Language.SystemVerilog.Parser.Parse
```

This works:
```
typedef struct packed
{
  logic [2:0] width;
} baz_s;

localparam baz_s my_baz = '{width: 2};

module foo
 #(parameter baz_s width_p = my_baz)
  (input [width_p-1:0] x
   , output [width_p-1:0] y
   );

   assign y = x + 1'b1;

endmodule
```
Result:
```
module foo (
	x,
	y
);
	// removed typedef: baz_s
	localparam [2:0] my_baz = '{
		width: 2
	};
	parameter [2:0] width_p = my_baz;
	input [(width_p - 1):0] x;
	output [(width_p - 1):0] y;
	assign y = (x + 1'b1);
endmodule
```

So it seems that sv2v doesn't like array of structs as parameters.

Page 85/86 are relevant here:
http://www.ece.uah.edu/~gaede/cpe526/2012%20System%20Verilog%20Language%20Reference%20Manual.pdf

The LRM doesn't explicitly forbid or permit structs as a type for parameters, but Synopsys tools _do_ accept them (and they're very useful in certain niche cases!).  Haven't checked with other vendors.

zachjs, 2019-08-07:
Thank you for another solid report!

Parameters are indeed explicitly allowed to have a `data_type`, and struct arrays fall under that definition. The relevant pages are 1145 and 1146 in IEEE 1800-2017, which I have been using as my SystemVerilog reference (a link is in the README).

I just pushed a commit which addresses this issue. Please let me know if it works for you!

dpetrisko, 2019-08-07:
Works! Thank you!

closed 2019-08-08 by zachjs