zachjs/sv2v #10: Support for packed unions

dpetrisko, 2019-08-08:
baz_pkg.v
```
package baz_pkg;

typedef union packed
{
  logic [5:0] a;
  logic [5:0] b;
} baz_u;

endpackage
```

Command:
```
sv2v baz_pkg.v
```

Result:
```
sv2v: Parse error: unexpected token 'packed' (KW_packed) at ./baz_pkg.v:4:15.
CallStack (from HasCallStack):
  error, called at .stack-work/dist/x86_64-linux/Cabal-2.4.0.1/build/sv2v/sv2v-tmp/Language/SystemVerilog/Parser/Parse.hs:6134:22 in main:Language.SystemVerilog.Parser.Parse
```

Packed unions are defined around page 142: IEEE_Std1800-2017

Thanks!

dpetrisko, 2019-08-08:
foo_pkg.v
```
package foo_pkg;

typedef union
{
  logic [5:0] a;
  logic [5:0] b;
} msg;

endpackage
```

Command:
```
sv2v foo_pkg.v
```

Result:
```
sv2v: Parse error: unexpected token '{' (Sym_brace_l) at ./foo_pkg.v:5:1.
CallStack (from HasCallStack):
  error, called at .stack-work/dist/x86_64-linux/Cabal-2.4.0.1/build/sv2v/sv2v-tmp/Language/SystemVerilog/Parser/Parse.hs:6134:22 in main:Language.SystemVerilog.Parser.Parse
```

For completeness, unpacked unions don't work either.  Unpacked unions are fairly worthless in general, though, since they're not typically synthesizable.

zachjs, 2019-08-09:
I've added support for parsing and converting packed unions. Let me know if it works for you!

dpetrisko, 2019-08-09:
Yep, seems to work!  Thanks!

closed 2019-08-09 by zachjs