zachjs/sv2v #22: Add switch to disable/enable defparam support

NilsGraf, 2019-08-29:
This spawned from issue #13. 

Most RTL coding guidelines ban defparam. The [SystemVerilog IEEE standard](http://ecee.colorado.edu/~mathys/ecen2350/IntelSoftware/pdf/IEEE_Std1800-2017_8299595.pdf) lists defparam as a candidate for deprecation in future versions (see Annex C.4.1), see also [here](https://blogs.mentor.com/verificationhorizons/blog/2010/03/23/the-art-of-deprecation/). 

Would it make sense to have a switch that enables/disables defparam support? Disabling defparam support would improve the readability of the generated Verilog, because parameters could be hardcoded (see #13 for examples of current Verilog output that supports defparam).

zachjs, 2019-08-30:
While eliminating defparams would be nice, the more likely scenario the ternary expressions handle is parameters being overridden via normal parameter bindings. I think this issue may be better framed at adding a switch that allows sv2v to make certain aggressive assumptions. 

For example, take a look at some of the output generated by converting https://github.com/zachjs/sv2v/blob/master/test/relong/array.sv:
```systemverilog
module Array (
	index,
	element,
	array,
	clock,
	clear,
	enable
);
	parameter ELEMENTS = 16;
	parameter WIDTH = 32;
	input wire [($clog2(ELEMENTS) - 1):0] index;
	input wire [(WIDTH - 1):0] element;
	output reg [(((ELEMENTS - 1) >= 0) ? (((WIDTH - 1) >= 0) ? (((((ELEMENTS - 1) >= 0) ? ELEMENTS : ((0 - (ELEMENTS - 1)) + 1)) * (((WIDTH - 1) >= 0) ? WIDTH : ((0 - (WIDTH - 1)) + 1))) + -1) : (((((ELEMENTS - 1) >= 0) ? ELEMENTS : ((0 - (ELEMENTS - 1)) + 1)) * ((0 >= (WIDTH - 1)) ? ((0 - (WIDTH - 1)) + 1) : WIDTH)) + ((WIDTH - 1) - 1))) : (((WIDTH - 1) >= 0) ? ((((0 >= (ELEMENTS - 1)) ? ((0 - (ELEMENTS - 1)) + 1) : ELEMENTS) * (((WIDTH - 1) >= 0) ? WIDTH : ((0 - (WIDTH - 1)) + 1))) + (((ELEMENTS - 1) * (((WIDTH - 1) >= 0) ? WIDTH : ((0 - (WIDTH - 1)) + 1))) - 1)) : ((((0 >= (ELEMENTS - 1)) ? ((0 - (ELEMENTS - 1)) + 1) : ELEMENTS) * ((0 >= (WIDTH - 1)) ? ((0 - (WIDTH - 1)) + 1) : WIDTH)) + (((WIDTH - 1) + ((ELEMENTS - 1) * ((0 >= (WIDTH - 1)) ? ((0 - (WIDTH - 1)) + 1) : WIDTH))) - 1)))):(((ELEMENTS - 1) >= 0) ? (((WIDTH - 1) >= 0) ? 0 : (WIDTH - 1)) : (((WIDTH - 1) >= 0) ? ((ELEMENTS - 1) * (((WIDTH - 1) >= 0) ? WIDTH : ((0 - (WIDTH - 1)) + 1))) : ((WIDTH - 1) + ((ELEMENTS - 1) * ((0 >= (WIDTH - 1)) ? ((0 - (WIDTH - 1)) + 1) : WIDTH)))))] array;
	input wire clock;
	input wire clear;
	input wire enable;
	localparam ZERO_ELEMENT = {WIDTH {1'b0}};
	always @(posedge clock) if (clear) array <= {ELEMENTS {ZERO_ELEMENT}};
	else if (enable) array[((((WIDTH - 1) >= 0) ? 0 : (WIDTH - 1)) + ((((ELEMENTS - 1) >= 0) ? index : (0 - (index - (ELEMENTS - 1)))) * (((WIDTH - 1) >= 0) ? WIDTH : ((0 - (WIDTH - 1)) + 1))))+:(((WIDTH - 1) >= 0) ? WIDTH : ((0 - (WIDTH - 1)) + 1))] <= element;
endmodule
```

While it works, this output is obviously not pretty. We intuitively know that `WIDTH` and `ELEMENTS` should never be negative, so the ternary expressions look very excessive. But sv2v isn't necessarily provided every instantiation of the module it is converting, so it can't know that parameters will, say, always have the same sign as their default value. If we allow sv2v to make that assumption, we get:
```systemverilog
module Array (
	index,
	element,
	array,
	clock,
	clear,
	enable
);
	parameter ELEMENTS = 16;
	parameter WIDTH = 32;
	input wire [($clog2(ELEMENTS) - 1):0] index;
	input wire [(WIDTH - 1):0] element;
	output reg [((ELEMENTS * WIDTH) + -1):0] array;
	input wire clock;
	input wire clear;
	input wire enable;
	localparam ZERO_ELEMENT = {WIDTH {1'b0}};
	always @(posedge clock) if (clear) array <= {ELEMENTS {ZERO_ELEMENT}};
	else if (enable) array[(0 + (index * WIDTH))+:WIDTH] <= element;
endmodule
```

One thing that would be nice to have regardless would be improved expression simplification. `(0 - (WIDTH - 1)) + 1` appears 10 times in the first output. It might not be too hard simplify it down to `2 - WIDTH`, for example.

NilsGraf, 2019-09-05:
Thanks, this makes sense. However, I've never seen a design that uses negative indices. Would it make sense to have a switch for enabling/disabling support for negative indices?

zachjs, 2019-09-14:
Even nonnegative indices could cause the direction of a range to flip. For example, consider `[X:4]` when X = 5 or 3. I haven't seen someone use parameters in such a way that it would flip the direction of a range, but perhaps it's not fair to assume that parameters always have reasonable default values.

Have the previous changes made the output cleaner? I'd be interested in seeing some complete output that is needlessly verbose so that we might come up with a more general solution.

NilsGraf, 2019-09-25:
I'm closing this issue, the generated Verilog seems fine (albeit it would be nice to have a switch for disabling support for negative indices).

closed 2019-09-25 by NilsGraf