zachjs/sv2v #23: [question] Is the grammar complete?

Nic30, 2019-08-31:
Hello,

I like simplicity of your [SV grammar](https://github.com/zachjs/sv2v/blob/master/src/Language/SystemVerilog/Parser/Parse.y)

Seeing from history, you probably building it on your own.
Do you have any system in it, I mean can you measure how much does remain?

If I compare for example property expression rule in this two grammars.
https://github.com/zachjs/sv2v/blob/master/src/Language/SystemVerilog/Parser/Parse.y#L498
https://github.com/Nic30/hdlConvertor/blob/master/grammars/sv2017Parser.g4#L662

There is wast difference. I would like to simplify the second grammar and I am asking if you do have some tool which can be used to simplify the grammar or you simplified some rules and you know that they are complete and correct.

Thanks for response,
Nic30

zachjs, 2019-08-31:
The grammar generally focusses on the synthesizable portions of SystemVerilog. It is also not very strict. There are plenty of things that the grammar accepts that are forbidden. The hope is that sv2v will not be given invalid input, as many users already have access to commercial tools like VCS for simulation. sv2v is meant to be a tool that enables _synthesis_ without commercial tools like Vivado. This is also why I have not written a type checker.

The same is true for the AST representation. While this is something I would like to cut down on in the future though, my primary concern for the AST is making the conversion phases easier to write. In general this values consistency.

The grammar in this project is written by hand. I did not use a tool for simplification. That ANTLR grammar looks consistent with the official specification, which is probably the best way to have a compliant parser. 

Some of the quirks of my grammar also arise from my use of `happy`, which generates LALR(1) parsers. For example, I have a sub-parser entirely for declarations. ANTLR may generate parsers which are not as limited.

Nic30, 2019-08-31:
Thanks for fast response, SV is full of legacy, hopefully you will not be forced to implement things like non ANSI corner cases and so on. I am still missing a tool for grammar optimisations.

As you did answered my question I am closing this issue.

closed 2019-08-31 by Nic30