zachjs/sv2v #24: New open source SystemVerilog compliance suite being created

mithro, 2019-09-03:
It might be of interest to you that the SymbiFlow project has started working on an open source System Verilog compliance suite which can be found at https://github.com/SymbiFlow/sv-tests

It's still very early stages, but its goal is to cover the whole of the System Verilog LRM. 

The project is currently trying to produce a "status report" for popular open source tools and your sv2v parser is being added in https://github.com/SymbiFlow/sv-tests/pull/47 . The resulting output report can be found at https://symbiflow.github.io/sv-tests/.

taylor-bsg, 2019-09-03:
> SymbiFlow/sv-tests#47 <https://github.com/SymbiFlow/sv-tests/pull/47> .

Super cool! It would be neat to run it on the commercial tools as well.
Also interesting would be a tool that you can run on a piece of source and
see what features it uses!

M

mithro, 2019-09-03:
@taylor-bsg Most of the commercial tools forbid you from publish this type of results and the symbiflow project only really cares about open source tooling. Nothing prevents you from running the test suite against a commercial tool yourself however.

taylor-bsg, 2019-09-03:
I think the open source community and commercial community alike would
benefit from more uniformity in support, since the side effect of such an
evaluation is that more code works in more places.

M

NilsGraf, 2019-09-04:
BTW, I'm in the process of testing this excellent sv2v tool on the [Ibex RTL](https://github.com/lowRISC/ibex) and a larger design. This includes formal verification by comparing the generated Verilog against the original SystemVerilog using a commercial LEC tool (such as [Conformal](https://www.cadence.com/content/cadence-www/global/en_US/home/tools/digital-design-and-signoff/logic-equivalence-checking/conformal-equivalence-checker.html) from Cadence).

zachjs, 2019-09-05:
@mithro That project looks like an incredible asset to the community. Thank you! I've already begun using it to find issues/features to add.

@NilsGraf I look forward to seeing the results! I've been plugging away at the many issues you've found. While I feel the tool is in a pretty good state (successfully converted, synthesized, and booted a RISC-V core), there is clearly more to be done. I appreciate the effort you've been putting in to help things along.

NilsGraf, 2019-09-25:
FYI, using latest sv2v version, I'm now able to convert Ibex SystemVerilog to Verilog and read it into Yosys without errors, see [this PR](https://github.com/lowRISC/ibex/pull/342). However, running Conformal LEC between original SystemVerilog and synthesized netlist doesn't match yet, still debugging it.

zachjs, 2019-10-05:
@NilsGraf I'm curious if you have any update on this. I'd be happy to help you with debugging in any way I can. If you have access to a commercial synthesis tool, it may be worth trying to disable different sets of conversions (beyond what the CLI provides) to narrow in on the source of issue.

zachjs, 2020-01-21:
I'm closing this issue for lack of activity. I will continuing using https://github.com/SymbiFlow/sv-tests to identify issues and missing support in sv2v. I've found it very helpful!

closed 2020-01-21 by zachjs