zachjs/sv2v #26: Package imports don't always work

NilsGraf, 2019-09-03:
File `foo_pkg.sv`:
```systemverilog
package foo_pkg;
  typedef enum logic [2:0] {
    AccessAck     = 3'd0,
    AccessAckData = 3'd1
  } inp_t;
endpackage
```
File `goo.sv`:
```systemverilog
module goo;
  import foo_pkg::*;
  inp_t test;
  assign test = foo_pkg::AccessAck;
endmodule
```
`sv2v -o foo_pkg.sv goo.sv` produces below, but it should also define parameter `foo_pkg_AccessAck`:
```systemverilog
module goo;
  wire [2:0] test;
  assign test = foo_pkg_AccessAck;
endmodule
```
Interestingly, if `goo.sv` is changed as follows, then correct output is generated:
Modified `goo.sv` (only line 4 is modified):
```systemverilog
module goo;
  import foo_pkg::*;
  inp_t test;
  assign test = AccessAck;
endmodule
```
Correctly generated Verilog (see line 2, which correctly defines the localparm 1AccessAck`):
```systemverilog
module goo;
  localparam [2:0] AccessAck = 3'd0;
  wire [2:0] test;
  assign test = AccessAck;
endmodule
```

NilsGraf, 2019-09-03:
Here is another testcase where `localparam AccessAck` is not generated. File `goo.sv`:
```systemverilog
module goo;
  import foo_pkg::*;
  wire [2:0] test;

  always_comb begin
    case (test)
      AccessAck: $display("Ack");
      default  : $display("default");
    endcase
  end
endmodule
```
Generated Verilog (it's missing definition of `AccessAck`):
```systemverilog
module goo;
  wire [2:0] test;
  always @(*) begin
    case (test)
      AccessAck: $display("Ack");
      default: $display("default");
    endcase
  end
endmodule
```

zachjs, 2019-09-05:
Enums in packages should be all set now. These test cases are very helpful. Thank you!

closed 2019-09-05 by zachjs

NilsGraf, 2019-09-05:
It also works on my side, thanks!

commit 41bba1b2 (2019-09-05): generate localparams for all package enums (closes #26)