zachjs/sv2v #28: Support for set membership operator "inside"

NilsGraf, 2019-09-03:
File `foo.sv`:
```systemverilog
module foo;
  logic [1:0] a;

  initial
    if (a inside {2'b01, 2'b11})
      $display("foo");

endmodule
```
Error message:
```
sv2v: Parse error: unexpected token 'inside' (Id_simple) at ./foo.sv:5:11.
CallStack (from HasCallStack):
  error, called at .stack-work/dist/x86_64-linux-tinfo6/Cabal-2.4.0.1/build/sv2v/sv2v-tmp/Language/SystemVerilog/Parser/Parse.hs:6348:22 in main:Language.SystemVerilog.Parser.Parse
```
`a inside {2'b01, 2'b11}` should be expanded to `( (a == 2'b01) || (a == 2'b11) )`.

zachjs, 2019-09-05:
Support for `inside` expressions has been added. Thanks for the report!

I don't yet have `case ... inside` support, though I haven't yet encountered its usage in the wild.

closed 2019-09-05 by zachjs

NilsGraf, 2019-09-05:
Thanks for the fast implementation, it works now fine on my side.

commit 1a6e0ce9 (2019-09-05): added support for inside expressions (closes #28)