zachjs/sv2v #39: Plans for Verilog attributes?

vegaluisjose, 2019-09-15:
Hi Zack,

Is there any plans for Verilog attributes?

For example, like [these](https://github.com/RedPitaya/RedPitaya/blob/master/fpga/rtl/red_pitaya_dfilt1.sv#L66)

Currently, the parser breaks under these.

zachjs, 2019-09-15:
Attributes are currently supported for module (and interface) items and procedural statements. The example you linked to parses fine, but was not being converted properly (the tagged declarations were left as `logic`s). I have gone ahead and pushed a fixed for that issue.

I am aware that attributes can be used in other locations, such as on expressions and on operators. I'd like to focus my efforts on supporting attributes where people actually use them. Which contexts do you need?

vegaluisjose, 2019-09-15:
I see, I am trying to run the compiler over this

```
(* use_dsp = "yes" *) module mult;
endmodule
```

And I am getting the following error

```
sv2v: Parse error: unexpected token '(*' (Sym_paren_l_aster) at ./example.v:1:1.
CallStack (from HasCallStack):
  error, called at .stack-work/dist/x86_64-linux/Cabal-2.4.0.1/build/sv2v/sv2v-tmp/Language/SystemVerilog/Parser/Parse.hs:6962:22 in main:Language.SystemVerilog.Parser.Parse
```

Attributes are used sometimes by HLS tools to annotate modules with constraints directly in the source code, for example map this to a DSP module, instead of using a separate file `tcl` for it.

zachjs, 2019-09-16:
I've added support for attributes on modules. Please feel free to open another issue if there is another scenario where you need attributes!

commit 5d80c830 (2019-09-16): support for module attributes (resolves #39)
closed 2019-09-16 by zachjs