zachjs/sv2v #45: Support "for (int i =0; ...)" inside always_comb blocks

NilsGraf, 2019-09-18:
```systemverilog
module foo;
  logic [3:0] goo;
  always_comb
    for (int i = 0; i < 4; i++)
      goo[i] = 1'b0;
endmodule
```
Generated Verilog:
```systemverilog
module foo;
	reg [3:0] goo;
	always @(*) begin : sv2v_autoblock_0
		reg signed [31:0] i;  // <-- issue: can't declare variable inside always block 
		for (i = 0; (i < 4); i = (i + 1))
			goo[i] = 1'b0;
	end
endmodule
```
Instead, it could generate something like this:
```systemverilog
module foo;
  reg [3:0] goo;
  integer i_autoblock_0;
  always @(*)  
    for (i_autoblock_0 = 0; (i_autoblock_0 < 4); i_autoblock_0 = (i_autoblock_0 + 1))
      goo[i] = 1'b0;
endmodule
```

zachjs, 2019-09-21:
I'm having trouble finding anything in the specifications about such declarations being forbidden inside of always blocks. I tested with both VCS (in Verilog mode) and iverilog, and neither complain when given the generated output. Can you help me understand why this is an issue?

NilsGraf, 2019-09-21:
You are right, it's apparently legal, which I didn't know. But Yosys doesn't seem to support this.

zachjs, 2019-09-22:
I'm surprised Yosys doesn't support this. Could you demonstrate exactly what issue you're running into? I have made a couple of small PRs for Yosys in the past. Perhaps this wouldn't be too hard to support.

NilsGraf, 2019-09-23:
Here the error message from Yosys, complaining about `reg signed [31:0] i;`:
```
ERROR:  syntax error, unexpected TOK_ID, expecting ',' or '=' or ';' or '['
```

I did some more experiments:

Yosys works fine if `reg signed [31:0] i;` is replaced by `integer i`, which is actually closer to the original SystemVerilog `for (int i, ...`.  Would it make sense if sv2v would use `integer` instead of `reg signed [31:0]` for loop variables?

NilsGraf, 2019-09-24:
I get` Yosys 0.8 (git sha1 5706e90)`

zachjs, 2019-09-24:
I have to say, I'm baffled. I can't reproduce the issue you're encountering. Are we running the same generated Verilog through Yosys?

```
$ yosys8 -V
Yosys 0.8 (git sha1 5706e908, clang 11.0.0 -fPIC -Os)
$ yosys8 foo.v
<SNIPPED>
-- Parsing `foo.v' using frontend `verilog' --

1. Executing Verilog-2005 frontend.
Parsing Verilog input from `foo.v' to AST representation.
Generating RTLIL representation for module `\foo'.
Successfully finished Verilog frontend.

End of script. Logfile hash: 543447362e, CPU: user 0.03s system 0.01s
Yosys 0.8 (git sha1 5706e908, clang 11.0.0 -fPIC -Os)
Time spent: 100% 1x read_verilog (0 sec)
$ cat foo.v
module foo;
	reg [3:0] goo;
	always @(*) begin : sv2v_autoblock_0
		reg signed [31:0] i;  // <-- issue: can't declare variable inside always block
		for (i = 0; (i < 4); i = (i + 1))
			goo[i] = 1'b0;
	end
endmodule
```

NilsGraf, 2019-09-25:
Ah, sorry, I copied the wrong testcase, below is the correct one (notice the `unsigned` loop variable). This testcase is derived from [this Ibex file](https://github.com/lowRISC/ibex/blob/master/rtl/ibex_register_file_ff.sv). 
```systemverilog
module foo;
  logic [3:0] goo;
  always_comb
    for (int unsigned i = 0; i < 4; i++)
      goo[i] = 1'b0;
endmodule
```
Generated Verilog:
```systemverilog
module foo;
	reg [3:0] goo;
	always @(*) begin : sv2v_autoblock_0
		reg unsigned [31:0] i; 
		for (i = 0; (i < 4); i = (i + 1))
			goo[i] = 1'b0;
	end
endmodule
```
Yosys:
```
yosys -QTqp "read_verilog foo.v"
ERROR: syntax error, unexpected TOK_ID, expecting ',' or '=' or ';' or '['
```
I get this Yosys error with both the old 0.8 version and the latest version.

zachjs, 2019-09-25:
Thanks for woking through this issue with me! It should now be fixed on master.

`reg` and `wire` are implicitly `unsigned`. Verilog-2005 doesn't allow the explicitly `unsigned` form. I have added a conversion which makes explicit `unsigned`s implicit, where possible.

closed 2019-09-25 by zachjs

NilsGraf, 2019-09-25:
Thanks for fixing it!

There is a similar issue related to unsigned parameters, for which I will file a new issue.

commit c7f51209 (2019-09-25): remove extraneous explicit unsigned (resolves #45)