zachjs/sv2v #48: Exclude array flattening feature

NilsGraf, 2019-09-25:
This is a request for the `--exclude=CONV` switch. Currently CONV includes `always, interface, logic`. It would be nice to also retain multi-dimensional arrays inside a module (not for interfaces, which is not supported by Verilog).

This would be helpful for register-files or memories. Flattening those to 1-dimensional regs makes it harder for synthesis tools to recognize them as memories.

zachjs, 2019-09-26:
I think sv2v is a bit overzealous with flattening arrays. I'd like rectify the behavior rather than add an option to exclude the problematic conversion.

Could you share your thoughts on the following options?
* Ideally, it would only flatten packed dimensions. However, any array ports must be packed and flattened. Arrays bound to such ports should presumably also be flattened.
* Another option would be to flatten everything except things like `reg [a:b] foo [c:d]`, though this becomes problematic if `foo` is a port.
* In the past (prior to 9a38225b1de25f3d2998dc2ff3f2e8fa1494a8b8), the conversion created flattened and unflattened versions of the original array such that each version could be used where needed. That procedure was complex, and never worked fully before I replaced it.
* I wonder if it's possible to continue aggressively flattening and have sv2v add an attribute that indicates to synthesis tools that the (converted) data declaration should be a memory based on it having been of the form `reg [a:b] foo [c:d]` prior to conversion.

NilsGraf, 2019-09-26:
All these options would work for me, but I'm not sure what kind of synthesis attributes / pragmas could be added for the last option.

An additional feature would be to convert packed arrays such as `reg [1:0][2:0][7:0] k;` to unpacked arrays  such as `reg [7:0] k [1:0][2:0];`, provided they are not used as ports.

zachjs, 2019-09-27:
This should now be fixed on master. I went with the first option. One extra scenario where I have it packing and flattening arrays is when they are assigned a value in a single assignment. Please let me know if you run into any issues with this change.

closed 2019-09-27 by zachjs

NilsGraf, 2019-09-27:
Thanks for the prompt implementation as always!

I have one question: I tried it out on [this ibex file](https://github.com/lowRISC/ibex/blob/master/rtl/ibex_cs_registers.sv) and noticed that some non-interface unpacked arrays are flattened, while others aren't (see below). Is this expected? 

Here are two unpacked arrays from [lines 175 and 176 of this ibex file](
https://github.com/lowRISC/ibex/blob/master/rtl/ibex_cs_registers.sv#L175-L176):
```systemverilog
  logic [63:0] mhpmcounter_mask [32];
  logic [63:0] mhpmcounter_d [32];
```
They are converted as follows (one is flattened, and one isn't):
```systemverilog
  logic [63:0] mhpmcounter_mask [0:31];
  logic [2047:0] mhpmcounter_d;
```

zachjs, 2019-09-28:
In that file `mhpmcounter_d` and `mhpmcounter_q` are both assigned values using single assignments: `mhpmcounter_d = mhpmcounter_q;` and `mhpmcounter_q <= mhpmcounter_d;`. Verilog does not allow this, so sv2v flattens the arrays. Given that those are procedural assignments, it would be possible to replace them with loops which assign each index. I think it would also be possible to replace them with concatenations such as `{a[2], a[1], a[0]} = ...`, which would work outside of procedural code, provided the bounds are not parameterized. I can look into these special cases if you think it would be worthwhile.

commit 454afa97 (2019-09-27): major array pack and flatten update (closes #48)

- dimensions flattening conversion only flattens packed dimensions
- conversion for packing arrays when necessary (preserves memories)
- added coverage for array with multiple packed dimensions
- test runner no longer forbids multi-dim accesses after conversion
- Decl and subordinate types derive Ord