zachjs/sv2v #50: Support for break keyword

Jbalkind, 2019-10-07:
When running on the issue unit of AnyCore, sv2v chokes on a break statement on the following line: https://github.com/anycore/anycore-riscv-src/blob/piton/issue/AgeOrdering.sv#L196

The error message is `./issue/AgeOrdering.sv:197:9: Parse error: unexpected token 'break' (KW_break).`

zachjs, 2019-10-08:
I will add support for jumping statements. Obviously the result may not be pretty, as Verilog does not have these constructs.

zachjs, 2019-10-09:
I've added support on master. The output isn't as clean as it could possibly be, but I am eager to find out if it works for you.

Jbalkind, 2019-10-10:
I'm not sure about correctness (and I probably won't have the chance to test it very soon) but with your most recent changes, sv2v now produces verilog for the whole of AnyCore!

zachjs, 2019-10-19:
I am closing this issue for the time being. If you find any issues with the conversion, please reopen this issue or file a new one.

closed 2019-10-19 by zachjs