zachjs/sv2v #55: Support for unique case inside

b1f6c1c4, 2019-11-19:
```systemverilog
module foo(
    input  logic x,
    output logic y
);

    always_comb
        unique case (x) inside
            [1:2]:   y = 0;
            default: y = 1;
        endcase

endmodule
```
Should translate to
```verilog
module foo (
        x,
        y
);
        input wire x;
        output reg y;
        always @(*)
                case (x)
                        1, 2: y = 0;
                        default: y = 1;
                endcase
endmodule
```

zachjs, 2019-12-02:
In the case of `[1:2]:`, this method of elaboration works easily enough. However, if the bounds of the range are parameters, I believe the case block would have to be elaborated into a series of `if` `else` blocks. Does this seem reasonable?

b1f6c1c4, 2019-12-02:
> In the case of `[1:2]:`, this method of elaboration works easily enough. However, if the bounds of the range are parameters, I believe the case block would have to be elaborated into a series of `if` `else` blocks. Does this seem reasonable?

Yes this is reasonable. It's also reasonable if sv2v transpiles every `unique case (...) inside` into `if/else`.

zachjs, 2019-12-22:
Sorry for the delay! Support has been added on master. The conversion does not yet support unpacked array patterns, and may evaluate the case expression multiple times (which violates the SystemVerilog specification). If you have the need, let me know and I can prioritize addressing these issues.

closed 2019-12-22 by zachjs