zachjs/sv2v #56: Wrapping if statements with begin/end

nmoroze, 2019-11-19:
I noticed some odd behavior when converting code with a snippet like this:
```systemverilog
always_comb begin
  x = 1'b0;
  y = 1'b0;

  if (a) begin
    if (b) begin
      x = 1'b1;
    end
  end else begin
    if (c) begin
      y = 1'b1;
    end
  end
end
```

sv2v doesn't wrap the body of the `if(a)` statement in `begin`/`end` tags, rather it gives output like this:
```verilog
always @(*) begin
  x = 1'b0;
  y = 1'b0;
  if (a)
    if (b)
      x = 1'b1;
  else if (c)
    y = 1'b1;
end
```

Although I initially assumed this was legal Verilog (since the body of the `if(a)` statement is just one statement itself), this doesn't seem to act as expected when running in simulation under both Icarus Verilog and Vivado. `y` doesn't get set to 1 even when `a` = 0, `b` = 0, and `c` = 1. However, adding the additional `begin` and `end` causes the code to work as intended.

I'm not 100% sure what the Verilog spec actually says here, but it may be a good idea to wrap all blocks in `begin` and `end` regardless of length to make sure all tools handle them correctly. 

Here's a more complete test case that should let you reproduce this behavior:
```systemverilog
`timescale 1ns / 1ps

module test();

logic a;
logic b;
logic c;

logic x;
logic y;

always_comb begin
  x = 1'b0;
  y = 1'b0;

  if (a) begin
    if (b) begin
      x = 1'b1;
    end
  end else begin
    if (c) begin
      y = 1'b1;
    end
  end
end

initial begin
  a = 1'b0;
  b = 1'b0;
  c = 1'b0;
  // y = 0 here
  #10;
  c = 1'b1;
  // expect y = 1 here, but it's still 0!
  #10;
  $finish;
end
endmodule
```

closed 2019-11-21 by zachjs

zachjs, 2019-11-21:
This is a great catch! Thank you for finding this. sv2v now wraps the "then" block of `if`/`else` statements with `begin`/`end` if that "then" block is a branching statement. This ensures that the else block associates as in the AST.

As a side note, in both C and SystemVerilog, `else` blocks associate to the most deeply nested matching `if`.

nmoroze, 2019-11-21:
> As a side note, in both C and SystemVerilog, else blocks associate to the most deeply nested matching if.

Great, thanks for clarifying that and for fixing the issue (and all the others I've opened) so quickly!

commit 35a0587d (2019-11-21): preserve else block association (resolved #56)