zachjs/sv2v #63: Parse error when a delay is used together with a macro

rbarzic, 2020-02-03:
The following code fails with a 

> Parse error: unexpected token ';' (Sym_semi).

when the INITIAL_VALUE macro is used

```verilog
`define INITIAL_VALUE 1'b0
module bug (data, clk );
   input clk;
   output reg data;

   always_ff @(posedge clk) begin
      data <= #1 `INITIAL_VALUE;      
   end
endmodule 

```
The following code is OK :

```verilog
`define INITIAL_VALUE 1'b0
module bug (data, clk );
   input clk;
   output reg data;

   always_ff @(posedge clk) begin
      data <= #1 1'b0;      
   end
endmodule 

```

This code is also OK
```verilog
`define INITIAL_VALUE 1'b0
module bug (data, clk );
   input clk;
   output reg data;

   always_ff @(posedge clk) begin
      data <= #1 `INITIAL_VALUE;      
   end
endmodule 
```
So the problem seems to be the combination of a delay with a macro.

(problem found when trying sv2v on code from https://github.com/pulp-platform/apb_i2c)

zachjs, 2020-02-05:
Thank you for spotting this issue and the test case! I'm working on a fix. This may take a couple days, as the issue is largely fundamental to the current architecture of the preprocessor.

zachjs, 2020-02-07:
This should now be fixed on master. Please let me know if you find any issues!

closed 2020-02-07 by zachjs

rbarzic, 2020-02-07:
Thanks  ! the latest version works perfectly