zachjs/sv2v #64: modport in interface

b1f6c1c4, 2020-02-05:
```systemverilog
interface I;
    logic f;
    modport P (input f);
endinterface

module M (I.P p);
    assign p.f = 1'b1;
endmodule

module MM;
    I inst();
    M a (.i(inst));
endmodule
```

Gives error:
```
sv2v: Map.!: given key is not an element in the map
CallStack (from HasCallStack):
  error, called at libraries/containers/Data/Map/Internal.hs:610:17 in containers-0.6.0.1:Data.Map.Internal
```

zachjs, 2020-02-08:
I believe there is an issue with your example. The instance `inst` is bound to a port with name `i`, but the module `M` only has a port named `p`. With new error messages I have added on master, sv2v now yields:
```
sv2v: could not find port "i" in module "M"
CallStack (from HasCallStack):
  error, called at src/Convert/Interface.hs:141:44 in main:Convert.Interface
```

Changing `M a (.i(inst));` to `M a (.p(inst));` gives the following:
```SystemVerilog
module M (p_f);
	input wire p_f;
	assign p_f = 1'b1;
endmodule
module MM;
	wire inst_f;
	M a(.p_f(inst_f));
endmodule
```

Am I mistaken? Can you provide an example which demonstrates the issue you found? I appreciate you taking the time to sort this out!

zachjs, 2020-02-14:
Do you have an update on this issue? I have fixed a couple issues with the interface conversion since you this was filed, so the original problem may already have been addressed.

b1f6c1c4, 2020-02-14:
The problem was that, if I put `interface I`, `module M`, `module MM` into three separate files,
calling `sv2v -I . MM.sv` will complain about missing M.
This is fixable by doing `sv2v -I . M.sv M.sv`.

However, since there are ~160 sv files in my project, it takes super long time to execute and the final output file is 15MiB. Any idea to accelerate the process?

zachjs, 2020-02-16:
I just pushed a change which should reduce the size of the output for files which contain imports at the global scope.

Can you share the source you are converting? If so, I would be happy to profile the execution. There may be some pattern your source exhibits that is handled poorly.

zachjs, 2020-02-21:
As another follow-up, I've just pushed some optimizations which reduced the runtime by 50% on some cores. Let me know if you see an improvement.

b1f6c1c4, 2020-02-21:
We are trying to convert [Ariane](https://github.com/pulp-platform/ariane) to verilog. Your newest fix is great and it takes only 4 minutes to convert the whole project. Thanks a lot!

closed 2020-02-21 by b1f6c1c4