zachjs/sv2v #65: Verbose version output

towoe, 2020-02-06:
Hi, thanks for the great tool!
This is a feature request for a more verbose output for the "--version" argument.
The current output I get:
```
sv2v v0.0.1, (C) 2019 Zachary Snow, 2011-2015 Tom Hawkins
```
From my perspective it would be great to include the git hash.
I have a `sv2v` binary inside a container (no more git sources) and for a quick check and comparison to maybe a local installation it would be helpful to see how far behind my version is from upstream/master.
What do you think @zachjs?

mithro, 2020-02-06:
`git describe` is a good tool for getting a nice version number which sorts nicely and includes the git hash. Just need to make sure you are using annotated tags and not doing a shallow clone.

towoe, 2020-02-06:
@mithro do you mean to use the output of `git describe` to be included in the `--version` output?
I would agree that this is an option, but `sv2v` does not really use version tags right now, so the plain hash has roughly the same expressiveness.

zachjs, 2020-02-08:
I have added a short git hash to the help text and version string:
```
sv2v v0.0.1 (eed5444)
```

Does this fulfill your need?

towoe, 2020-02-08:
Thanks a lot @zachjs.
https://github.com/zachjs/sv2v/commit/eed5444d4a44f65764f0b4c1660706181b32900c does exactly what I meant!

closed 2020-02-08 by towoe