zachjs/sv2v #66: Parsing defined macros

wtatarski, 2020-02-12:
I'm trying to parse file https://github.com/lowRISC/ibex/blob/master/shared/rtl/ram_1p.sv with the following commands:
`sv2v -D SRAM_INIT_FILE=examples/sw/led/led.vmem shared/rtl/ram_1p.sv`

It fails on this assignment:
```localparam MEM_FILE = `"`SRAM_INIT_FILE`";```

with error:
`shared/rtl/ram_1p.sv:78:28: Lexical error: Undefined macro:`

Is it correct way of passing macros?

wtatarski, 2020-02-12:
Actually when I changed the wrong line to 
```localparam MEM_FILE = "`SRAM_INIT_FILE";``` 
there is no error. However in that case MEM_FILE parameter is not bound correctly.

zachjs, 2020-02-13:
Your original code is the correct insofar as most tools (commercial or otherwise) will exhibit the behavior you're expecting. Interestingly, the IEEE 1800-2017 doesn't appear to provide a specification for this behavior.

I think the confusion comes from section 22.5.1 (the specification for ``` `define ```), which says:
> An ``` `" ``` overrides the usual lexical meaning of `"` and indicates that _the expansion_ shall include the quotation mark, substitution of actual arguments, and expansions of embedded macros. This allows string literals to be constructed from macro arguments.

I would interpret this to only apply to _macro expansions_, and not elsewhere in the source. However, many tools interpret ``` `" ``` outside of macro expansions similar to how they are handled within macro expansions. In general, these strings can only be terminated by a matching ``` `" ```. Within them, macro substitutions are performed and backticks must be escaped. Behavior and support varies significantly for more complex uses of this syntax.

Despite the ambiguity, I have implemented and pushed changes to master to support such "backtick string literals" outside of macro definitions. The behavior is now largel in line with that of VCS. If sv2v is still doing something silly, or if I have missed some relevant portion of the specification, please let me know!

wtatarski, 2020-02-13:
Looks good to me. 
Thanks @zachjs, your commit solved my issue.

closed 2020-02-13 by wtatarski