zachjs/sv2v #67: Invalid conversion of signals

wtatarski, 2020-02-13:
Input .sv code:
```
module registers;

  logic [63:0] mhpmcounter_d [32];
  logic [63:0] mhpmcounter [32];

  always_comb begin : mhpmcounter_update
    mhpmcounter_d = mhpmcounter;
  end

endmodule
```

Output verilog code:
```
module registers;
	reg [2047:0] mhpmcounter_d;
	wire [63:0] mhpmcounter [0:31];
	always @(*) begin : mhpmcounter_update
		mhpmcounter_d = mhpmcounter;
	end
endmodule
```

Then in Yosys I got error `ERROR: Invalid array access.` which is quite obvious.

Originally, it happened to me when parsing this [file](https://github.com/lowRISC/ibex/blob/master/rtl/ibex_cs_registers.sv#L186)

zachjs, 2020-02-14:
Thank you for finding this issue! It should now be fixed on master.

wtatarski, 2020-02-14:
Thank you for your fix! LGTM

closed 2020-02-14 by wtatarski