zachjs/sv2v #68: Invalid conversion of array indexes?

wtatarski, 2020-02-14:
.sv code:
```
module registers();

  logic [63:0] mhpmcounter_d [32];
  logic [63:0] mhpmcounter [32];

  always_comb begin : mhpmcounter_update
    mhpmcounter_d = mhpmcounter;

    for (int i=0; i<32; i++) begin : gen_mhpmcounter_update
      if (mhpmcounter_we[i]) begin
        mhpmcounter_d[i][31:0] = csr_wdata_int;
      end else if (mhpmcounterh_we[i]) begin
        mhpmcounter_d[i][63:32] = csr_wdata_int;
      end
    end

  end
endmodule
```
Output verilog code:
```
module registers;
	reg [2047:0] mhpmcounter_d;
	wire [2047:0] mhpmcounter;
	always @(*) begin : mhpmcounter_update
		mhpmcounter_d = mhpmcounter;
		begin : sv2v_autoblock_3
			reg signed [31:0] i;
			for (i = 0; (i < 32); i = (i + 1))
				begin : gen_mhpmcounter_update
					if (mhpmcounter_we[i])
						mhpmcounter_d[((31 - i) * 64)+:32] = csr_wdata_int;
					else if (mhpmcounterh_we[i])
						mhpmcounter_d[(((31 - i) * 64) + 32)+:32] = csr_wdata_int;
				end
		end
	end
endmodule
```

When `i=0`, index of `mhpmcounter_d` array is set to access last 32 bits. Is it right? Shouldn't it access first 32 bits? Is this `31 - ` required?

zachjs, 2020-02-15:
According to the specification, "`[size]` becomes the same as `[0:size-1]`." This means that the layout of `logic [63:0] mhpmcounter_d [32];` is not continuous as one might expect.

When sv2v packs and flattens arrays, it preserves this layout. Although the layout may not be consequential in the module you gave, this is not always the case. For example, when I run
```SystemVerilog
module top;
    logic [2:0] a [2];
    logic [2:0] b [1:0];
    initial begin
        $monitor("%b %b %b %b", a[0], a[1], b[0], b[1]);
        #1 a[0][0] = 1;
        #1 b[0][0] = 1;
        #1 a = b;
    end
endmodule
```
through VCS, I get
```
xxx xxx xxx xxx
xx1 xxx xxx xxx
xx1 xxx xx1 xxx
xxx xx1 xx1 xxx
```

sv2v preserves the layout of the arrays essentially by flipping around the indices of accesses according to the orientation of array dimensions. So for this example, we get
```SystemVerilog
module top;
	reg [5:0] a;
	reg [5:0] b;
	initial begin
		$monitor("%b %b %b %b", a[3+:3], a[0+:3], b[0+:3], b[3+:3]);
		#(1) a[3] = 1;
		#(1) b[0] = 1;
		#(1) a = b;
	end
endmodule
```

wtatarski, 2020-02-17:
Thank you for clarifying this @zachjs

closed 2020-02-17 by wtatarski