zachjs/sv2v #71: "-:" part select on multi-bit struct fields expands into invalid indexing

nmoroze, 2020-02-14:
The following input: 
```systemverilog
module test(

);

typedef struct packed {
    logic [7:0] field;
} struct_t;
struct_t mystruct;

reg [3:0] out;

assign out = mystruct.field[7-:4];

endmodule
```

Gives the following output:
```verilog
module test;
        wire [7:0] mystruct;
        wire [3:0] out;
        assign out = mystruct[7:0][7-:4];
endmodule
```

Which is illegal since you can't chain bit-selections like that. This only seems to happen with the "-:" operator in particular, since "+:" seems to produce legal output (combining the two ranges to produce a single set of proper indices).

zachjs, 2020-02-17:
Thank you for spotting and reporting this issue! This and a few adjacent issues have now been fixed in 8f2d7dd5c72ae5d40d2c61460c00973d5afe691e. I'm leaving this open pending your confirmation.

zachjs, 2020-02-17:
Also worth noting is that the updated conversions may potentially raise (innocuous) warnings in Yosys due to #1649/#1710.

nmoroze, 2020-02-18:
Thanks for the quick response! I think 8f2d7dd may have created a new issue unfortunately - it seems now that sv2v crashes when trying to convert inputs that assign struct fields with parameterized dimensions. Here's a fairly minimal example input:
```systemverilog
module test ();

localparam N = 5;

typedef struct packed {
    logic [N:0] field;
} mystruct;

mystruct a;

assign a.field = 5'b0;
endmodule
```

This causes sv2v to fail with the error:
`sv2v: src/Convert/Struct.hs:257:17-38: Non-exhaustive patterns in Just l'`

zachjs, 2020-02-19:
Thank you for catching and reporting that issue! It should now be resolved on master. Any futures instances of such bugs in those conversions should also produce much more useful error messages.

nmoroze, 2020-02-19:
Great, that's fixed now! Unfortunately I think I've caught another issue with this fix - now, it seems like if you have an array of structs nested inside another struct, references to fields of the nested struct aren't converted. Sample input:
```systemverilog
module test (
    output [3:0] out
);

typedef struct packed {
    logic [3:0] field;
} another_struct;

typedef struct packed {
    another_struct [3:0] nested;
} mystruct;

mystruct a;

assign out = a.nested[0].field;

endmodule
```

Output:
```verilog
module test (out);
        output [3:0] out;
        wire [15:0] a;
        assign out = a[0+:4].field;
endmodule
```

zachjs, 2020-02-20:
Thank you for continued feedback! This was a silly bug on my part. It's fixed in a415d9eb3d4a30bd6236186a6f64259952369e94.

nmoroze, 2020-03-04:
Fix is working for me, thank you! Sorry for the delay checking this time around.

closed 2020-03-04 by nmoroze