zachjs/sv2v #74: Parameter integer array isn't converted with explicit widths

nmoroze, 2020-02-15:
If you have a parameter array like this:
```systemverilog
module test();
localparam int unsigned params [4] = { 1, 2, 3, 4 };
endmodule
```

sv2v generates Verilog output like:
```verilog
module test;
        localparam [127:0] params = {1, 2, 3, 4};
endmodule
```

Vivado will accept this with a warning (and it will infer a width of 32 bits), but this causes Icarus to output an error for each value - "error: Concatenation operand "'sd1" has indefinite width." I think it would be best to specify the widths explicitly, i.e.:
```verilog
module test;
        localparam [127:0] params = {32'd1, 32'd2, 32'd3, 32'd4};
endmodule
```

zachjs, 2020-02-17:
This conversion is now handled on master. Please let me know if you run into any issues!

closed 2020-02-17 by zachjs

nmoroze, 2020-02-18:
Thanks!

I just realized that there's a separate but related edge case (turns out it was what I had come across originally) that still isn't handled: doing a similar assignment but with an assignment pattern instead of a concatenation. For example, this (note the extra `'`):
```systemverilog
module test();
parameter int signed params [4] = '{ 1, 2, 3, 4 };
endmodule
```
still gets converted without explicit widths.

zachjs, 2020-02-20:
Thank you for pointing this out! This case should also now be covered (as of 470fa01eb2aa1c80c70194a85262ba4d3e7bb3a7). For greater visibility, please feel free to re-open issues if you find they haven't been fully resolved.

nmoroze, 2020-03-04:
Fixed for me now!