zachjs/sv2v #75: Seeking feedback: binary distributions

zachjs, 2020-02-24:
Thanks in large part to the work of @hofstee, sv2v now has binary distributions that are automatically uploaded to the [releases page](https://github.com/zachjs/sv2v/releases). I am looking for feedback on this system. Please share your thoughts!

**Questions:**
- Do you have a need for 32 bit Linux or macOS builds?
- Is there interest in a Windows binary?
- How do you want to procure binaries?
  - @Kuree demonstrated that sv2v could be released on PyPi.
  - I have a little bit of experience with Homebrew.
  - What about other package managers?

Kuree, 2020-02-25:
Here are my thoughts on this
> Do you have a need for 32 bit Linux or macOS builds?

Popular Linux distro such as Ubuntu has dropped 32-bit OS support years ago. From maintenance perspective, I don't think providing 32-bit builds worths the effort.

> Is there interest in a Windows binary?

Most commercial EDA tools related to RTL synthesis and simulation only run on Linux, with the exception of Vivado, which also runs on Windows. Since yosys has official support on Windows, I think it makes sense to support Windows.

> How do you want to procure binaries
- Linux:
  Given how fragmented Linux package managers are, I think the best way to distribute is through PyPi. Since [PEP 599](https://www.python.org/dev/peps/pep-0599/) imposes a strict policy on the binary format, the package should work across the majority of Linux distros (hence the name "manylinux"). In addition, Python packages can be installed without root permission, which is another advantage.
  If you want more native distribution, we need to have distribution methods for each major distros. We can maintain a custom PPA for Ubuntu-based distros, and openSUSE for Fedora and SUSE. However, it is just more work in my opinion.
- macOS:
  IMO the best platform to distribute is homebrew. Almost every CI that supports macOS has homebrew installed.
- Windows:
  Windows has several popular package distribution systems such as [chocolatey](https://chocolatey.org/docs/create-packages). I think both appveyor and Github Actions have chocolatey installed by default. I'm not familiar with uploading custom packages to chocolatey, but I think it's worth a try.
 Another approach would be leveraging PyPI by making a Python package with proper Windows tag.

nmoroze, 2020-03-04:
I'll probably continue building from source for my own use so that I can stay up-to-date with master (and it's fairly easy since I have the toolchain set up already). 

However, if I were to use prebuilt binaries - personally I only use sv2v on Ubuntu and MacOS, so apt and Homebrew would be my picks. I don't need 32-bit builds!

zachjs, 2020-03-19:
@Kuree Thanks for pointing out the Conda package. I put in a request to add the `NOTICE` file to the distribution. Does distribution via Conda generally satisfy your needs?

Kuree, 2020-03-19:
Although conda is not shipped with most Linux distributions, it is fairly easy to install, especially in a CI environment. I'm quite okay with the conda package.

mithro, 2020-03-20:
@zachjs + @Kuree -- Hi from the @SymbiFlow project! We are currently undergoing a project to expand the SymbiFlow conda packages to support all of Linux (including RPi), Windows and Mac. We will most certainly include zachjs-sv2v as a tool there.

towoe, 2020-03-23:
Not sure what would be the best way to provide prebuilt binaries but I think it is good to have it. The setup of Haskell + Stack might be easy, but is from my perspective quite time consuming. For users of other projects who only want to make use of sv2v this adds an additional layer of complexity.

zachjs, 2020-03-31:
Do any of you use sv2v on a Windows machine? I have made a Windows binary available in new releases: https://github.com/zachjs/sv2v/releases/tag/v0.0.3.

zachjs, 2020-06-17:
I'm closing this now that binary release have been available for some time. If anyone runs into issues with them, please let me know!

closed 2020-06-17 by zachjs