zachjs/sv2v #76: Feature request: keeping `ifdef?

cpitclaudel, 2020-03-05:
Hi there,

Is there a way (or would it be easy to implement?) to keep `ifdef` maros in the generated Verilog?  My use case if that I have `ifdef` macros to chose between synthesis and simulation code, and I make that decision after convertion to Verilog; ideally, I'd rather not have to generate two copies of the Verilog file from the SystemVerilog one.

Thanks for the great work.

de0u, 2020-03-05:
Potentially relevant: https://dotat.at/prog/unifdef/

cpitclaudel, 2020-03-05:
I think this does the reverse f what I want, doesn't it?

zachjs, 2020-03-08:
What aspect of generating two copies of Verilog output hinders your specific use case? Would it be possible for you to share the source you are converting?

The ability for `ifdef` branches to affect the semantics of other constructs makes them not trivially representable in an AST. Consider the operator precedence in this trivial example:
```systemverilog
assign x = a - b
`ifdef FOO
    *
`else
    +
`endif
    c;
```

The most likely way I would try implementing this within sv2v would be to convert the source into two separate results and merge them using a diffing algorithm. However, this may be beyond the scope of the tool. `diff` already supports supports merging files using C-style preprocessor directives. Try using:
```sh
diff --ifdef=FOO a.v b.v | sed -e "s/^#/\`/"
```

zachjs, 2020-03-21:
@cpitclaudel Do you have an update on this request?

cpitclaudel, 2020-03-21:
Ouch, sorry for missing your previous comment.  You make an excellent point: this is hard.   And I think diff --ifdef is my best bet, indeed (I didn't know about that feature) :) Thanks a lot!

closed 2020-03-21 by cpitclaudel