zachjs/sv2v #80: Feature request: Support for net strength values

rw1nkler, 2020-03-20:
Hi, 

I have found that strength values for nets are not supported. They are specified in the [IEEE 1800-2017](https://ieeexplore.ieee.org/servlet/opac?punumber=8299593) at point 6.7, page 97.

Short description can be found here:
https://www.hdlworks.com/hdl_corner/verilog_ref/items/Strengths.htm

Code example:

```systemverilog
module top();

wire a, b, c;
assign (strong0, strong1) c = a + b;
endmodule
```

Returns the following error:
```
module.sv:4:10: Parse error: unexpected token '(' (Sym_paren_l).
```

The module can be successfully synthesized with Vivado.

zachjs, 2020-03-21:
I've added language support for these attributes on master. Please let me know if it works for you.

rw1nkler, 2020-03-23:
Thank you! Yes, It is working!

closed 2020-03-24 by zachjs