zachjs/sv2v #81: Packed array access

towoe, 2020-03-22:
Following packed slice access in https://github.com/zachjs/sv2v/commit/c03dba096f4d8c832228668685484174546d31a2 I think the ordering of the access is incorrect for accessing a single element.

Looking at a slightly modified test:
```diff
diff --git a/test/basic/multipack_port.sv b/test/basic/multipack_port.sv
index f19f73c..b143e19 100644
--- a/test/basic/multipack_port.sv
+++ b/test/basic/multipack_port.sv
@@ -28,8 +28,10 @@ module top;
     Producer #(.INIT(0)) p1(clock, foo);
 
     logic [10:0] bar [10];
-    Producer #(.INIT(0)) p2(clock, bar[0:4]);
-    Producer #(.INIT(1)) p3(clock, bar[5:9]);
+    Producer #(.INIT(0)) p2(clock, bar[0:2]);
+    Producer #(.INIT(1)) p3(clock, bar[3]);
+    Producer #(.INIT(1)) p4(clock, bar[4:4]);
+    Producer #(.INIT(2)) p5(clock, bar[5:9]);
 
     initial
         $monitor("%d %b%b%b%b%b %b%b%b%b%b%b%b%b%b%b", $time,
```

Will give the following part selections:
```verilog
		bar[32-:33]
		bar[66+:11]
		bar[54-:11]
		bar[109-:55]
```
The second access should probably be `bar[33+:11]`, if using the down vector with the base at LSB.

(And the monitor access shows `bar[99+:11], bar[88+:11], bar[77+:11], bar[66+:11], bar[55+:11], bar[44+:11], bar[33+:11], bar[22+:11], bar[11+:11], bar[0+:11]`)

So my assumptions is that somewhere the order needs to be switched.
I would like to do a PR, but I am totally unfamiliar with Haskell.

@rswarbrick for reference.

zachjs, 2020-03-22:
There was indeed a bug regarding the direction of converted range accesses of multi-dimensional arrays. Additionally, c03dba0 made the `multipack_port` test invalid. Thank you for bringing this to my attention!

Please let me know if you find any further issues!

zachjs, 2020-03-22:
There was indeed a bug regarding the direction of converted range accesses of multi-dimensional arrays. Additionally, c03dba0 made the `multipack_port` test invalid. Thank you for bringing this to my attention! I have fixed the test and expanded it to cover the cases you highlighted. The conversion behavior is also now corrected on master.

I don't believe that `Producer #(.INIT(1)) p3(clock, bar[3]);` is valid, as. VCS, for one, rejects it due to a type mismatch. I used `bar[3:3]` instead. Additionally, I had to add a parameter get the widths to match up in each instantiation.

Please let me know if you find any further issues!

towoe, 2020-03-23:
Thanks a lot for the prompt fix!

> I don't believe that `Producer #(.INIT(1)) p3(clock, bar[3]);` is valid, as. VCS, for one, rejects it due to a type mismatch. I used `bar[3:3]` instead. Additionally, I had to add a parameter get the widths to match up in each instantiation.
> 
Sorry, haven't really checked for the module's port validity, only the generated part selection.
Actually I am not sure what would be the correct definition in order to use `bar[3]`, but it seems a bit strange that a slice is necessary if the sizes match.

closed 2020-03-23 by towoe