zachjs/sv2v #82: Faulty iteration through two dimensional array

wtatarski, 2020-03-24:
sv code 
```
module alert_handler_class (
  input [3:0][1:0] loc_alert_class_i
);

  logic [3:0][3:0] loc_class_masks;

  always_comb begin : p_class_mask
    loc_class_masks = '0;
    for (int unsigned kk = 0; kk < 4; kk++) begin
      loc_class_masks[loc_alert_class_i[kk]][kk] = 1'b1;
    end
  end

endmodule : alert_handler_class
```

output verilog code:
```
module alert_handler_class (loc_alert_class_i);
	input [7:0] loc_alert_class_i;
	reg [15:0] loc_class_masks;
	always @(*) begin : p_class_mask
		loc_class_masks = 1'sb0;
		begin : sv2v_autoblock_2
			reg [31:0] kk;
			for (kk = 0; kk < 4; kk = kk + 1)
				loc_class_masks[(loc_alert_class_i[2 * (kk * 2)+:4] * 4) + kk] = 1'b1;
		end
	end
endmodule
```

I suspect that conversion of looping through two dimensional array is not correct. I got errors and warnings that selected range is out of bounds for signal `loc_alert_class_i` in that case.

zachjs, 2020-03-25:
Thank you for filing this issue! It is now fixed on master.

You've found a pretty interesting bug here.

The AST distinguishes between "left-hand sides" and "expressions" (or right-hand sides). Because LHSs are a subset of RHSs, the array flattening conversion would turn the each top-level LHS into an RHS, convert the RHS and any RHSs it contained, and then convert the result back to an LHS.

This caused an issue because the LHS `loc_class_masks[loc_alert_class_i[kk]][kk]` _contains_ the RHS `loc_alert_class_i[kk]`. This caused the conversion to attempt to flatten that usage of `loc_alert_class_i` twice in one pass.

The conversion now performs the traversal on the top-level LHSs and performs the LHS->RHS->convert->LHS procedure at each node.

wtatarski, 2020-03-25:
Thank you for your fix. It works!

closed 2020-03-25 by wtatarski