zachjs/sv2v #83: Feature request: Support for empty module parameter port list

rw1nkler, 2020-03-25:
Hey,

SystemVerilog requires module parameter port list to be present ([IEEE 1800-2017](https://ieeexplore.ieee.org/servlet/opac?punumber=8299593), point 23.2.2, page 697).
Currently, the empty parameter port list is not supported.

The following SystemVerilog code:
```systemverilog
module top
#() (
    input in,
    output out
);

    assign out = in;
endmodule
```
Causes the following error:
```
module.sv:2:3: Parse error: unexpected token ')' (Sym_paren_r).
```

The module can be synthesized using Vivado or Yosys.
The corresponding section for Verilog module parameter port list can be found in the [IEEE 1364-2005](https://ieeexplore.ieee.org/servlet/opac?punumber=10779), Annex A.1.3, page 487.

zachjs, 2020-03-26:
Thank you for filing this! I wasn't aware that the `parameter_port_list` could be empty. It's now allowed on master.

closed 2020-03-26 by zachjs

nilswiersma, 2020-03-30:
A similar issue occurs when instantiating a module , ie.
`top1 #( ) top (.*);`
results in 
` Parse error: unexpected token ')' (Sym_paren_r).`

zachjs, 2020-03-31:
@nilswiersma That should now be fixed as of bcafef8d01413c26281e1d6767683a89c74ac449.

commit 1687b1c5 (2020-03-26): allow empty parameter_port_list (resolves #83)