zachjs/sv2v #84: Can't build as a submodule

dpetrisko, 2020-04-05:
With the new build system, I'm able to build sv2v standalone but not when I have sv2v as a submodule.

```
<src_dir>/sv2v/src/Job.hs:35:18: error:
    • Exception when trying to run compile-time code:
        GHECouldn'tReadFile "<src_dir>/sv2v/.git/HEAD" <src_dir>sv2v/.git/HEAD: openBinaryFile: inappropriate type (Not a directory)
      Code: tGitInfoCwd
    • In the Template Haskell splice $$tGitInfoCwd
      In the first argument of ‘giHash’, namely ‘$$tGitInfoCwd’
      In the expression: giHash $$tGitInfoCwd
   |
35 | gitHash = giHash $$tGitInfoCwd
   |                  ^^^^^^^^^^^^^
[12 of 65] Compiling Language.SystemVerilog.AST.ModuleItem[boot]
```

OS: CentOS 7
Git: 2.16.5
Stack: 2.1.3

zachjs, 2020-04-05:
This appears to be related to https://github.com/snoyberg/githash/issues/16.

zachjs, 2020-04-05:
This appears to have already been fixed in that dependency, but I was using an outdated version. Upgrading the stackage resolver fixed it locally for me. Can you try the latest master (543a1046833e11507e7379cabc9c60b1377850fd)?

dpetrisko, 2020-04-05:
Thanks, that fixed it!

closed 2020-04-05 by dpetrisko