zachjs/sv2v #86: Bug when module names contain underscores

b1f6c1c4, 2020-05-31:
Given input:
```systemverilog
package PKG;
        localparam P = 2;
        typedef struct packed {
                logic [P-1:0] s;
        } S;
endpackage
module F_1 #(parameter type T)();
        F_2 #(.T(T)) inst ();
endmodule
module F_2 #(parameter type T = logic)();
        T t;
endmodule
module AUX ();
        F_1 #(.T(PKG::S)) inst ();
endmodule
```

sv2v output:
```verilog
module F_1_8571F;
        localparam PKG_P = 2;
        F_2_sv2v_default_827B5 inst();
endmodule
module F_2_sv2v_default_827B5;
        wire [((PKG_P - 1) >= 0 ? PKG_P : 2 - PKG_P) - 1:0] t;
endmodule
module F_2_AB763;
        wire t;
endmodule
module AUX;
        F_1_8571F inst();
endmodule
```

Problem: `F_2_sv2v_default_827B5` mistakenly treat `T` as `PKG::S` instead of `logic`, causing illegal syntax error (`PKG_P` not found).

zachjs, 2020-06-01:
While the name of the modules was not a factor, there were a couple of issues at play here.

Param type modules were being incorrectly instantiated. `F_2_sv2v_default_827B5` was clearly not the "default" instantiation of `F_2`. Fixing this exposed an issue where multiple instances of the same name could be created, but this too has been fixed.

The primary issue, though, was that sv2v had no support for type parameters which themselves relied on other parameters. To support this in what I hope is the typical case, I have made sv2v substitute in the value of localparams which are used in module instantiation type parameter bindings.

The output for your given example is now:
```systemverilog
module F_1_8571F;
	localparam PKG_P = 2;
	F_2_95780 inst();
endmodule
module F_2_AB763;
	wire t;
endmodule
module F_2_95780;
	wire [1:0] t;
endmodule
module AUX;
	F_1_8571F inst();
endmodule
```

b1f6c1c4, 2020-06-01:
The fix sounds nice. Thx for the help & closing issue now.

closed 2020-06-01 by b1f6c1c4