zachjs/sv2v #89: Discussion: What should be the correct way to handle initial statements

b1f6c1c4, 2020-06-01:
I'm wondering if the current way of handling `initial`, which is to convert everything to `initial ;`, is proper or not. I'm more inclined to remove them completely but I don't know if there will be any negative effect.

zachjs, 2020-06-01:
There isn't an explicit rule which replaces `initial` blocks with the empty `initial` blocks (i.e., `initial ;`).  However, it is possible for the conversion procedure to produce an empty block in certain circumstances. My guess is that you have something like:
```SystemVerilog
module top;
    typedef logic [16:1] T;
    initial begin
        if ($bits(T) % 8 != 0) $display("FOO");
    end
endmodule
```
which, because the `if` statement is elaborated and removed, converts to
```verilog
module top;
	initial ;
endmodule
```
With a different condition, that initial block would not end up empty.

Are you requesting that empty `initial` blocks be removed to clean up the output? This would be a very straightforward change.

b1f6c1c4, 2020-06-02:
I'm considering removing empty `initial` blocks simply because some tools like (VCS) produce warnings on that. Too many warnings are never beneficial.

zachjs, 2020-06-02:
I pushed a change which removes empty `initial` blocks. Let me know if you find an issue!

commit ec760964 (2020-06-02): remove empty initial blocks (resolves #89)
closed 2020-06-02 by zachjs