zachjs/sv2v #90: Bug when passing a local struct to parameter type with field referencing a parameter

b1f6c1c4, 2020-06-01:
This is also an edge-case bug related to `parameter type`.

Input:
```systemverilog
package PKG;
	typedef struct packed {
	        int W;
	} F_t;
	localparam F_t F_v = '{
	        W: 64
	};
endpackage
module M0 #(parameter type T) ();
	T v;
endmodule
module M1 #(parameter PKG::F_t F = PKG::F_v) ();
	typedef struct packed {
	        logic [F.W-1:0] field;
	} local_t;
	local_t v;
	M0 #(.T(local_t)) inst ();
endmodule
```
Output:
```verilog
module M0_876C2;
	wire [((F.W - 1) >= 0 ? F.W : 2 - F.W) - 1:0] v;
endmodule
module M1;
	localparam [31:0] PKG_F_v = sv2v_struct_70EB4(64);
	parameter [31:0] F = PKG_F_v;
	wire [((F[31-:32] - 1) >= 0 ? F[31-:32] : 2 - F[31-:32]) - 1:0] v;
	M0_876C2 inst();
	function automatic [31:0] sv2v_struct_70EB4;
	        input reg signed [31:0] W;
	        sv2v_struct_70EB4 = W;
	endfunction
endmodule
```
The module `M0_876C2` contains syntax error that `F` is not defined.

b1f6c1c4, 2020-06-02:
This is still the case on the newest `sv2v v0.0.3 (478f0d1)`.

zachjs, 2020-06-02:
Adding support for type parameters which themselves depend on other parameters isn't entirely trivial. At a high level, I'll need to add new parameters and bindings to the instantiations, and rewrite the type parameters to depend on these new parameters.

If you change `parameter PKG::F_t F` to a `localparam`, sv2v will handle it correctly. Does this suffice for your needs?

b1f6c1c4, 2020-06-02:
Well using `localparam` can be a decent workaround. However `M1` here is a top-level module that is to be included in another project, while `PKG:F_t F` is its configuration. Thus the `localparam` workaround may need more modification than expected. I think such a configuration parameter may be quite common in large designs. If possible, I'm still looking forward to the support. *But take your time, no need to hurry.*

zachjs, 2020-07-09:
Sorry for the delay in addressing this! I've added support for type parameters which themselves depend on other parameters in 671101a30b64575af7a21500cd7226c89d886c06. Please let me know if it works for you!

b1f6c1c4, 2020-07-10:
Unfortunately, this issue should be reopened because I noticed the following behavior on 
```
> sv2v --version
sv2v v0.0.4 (a83cc38)
```

Input:
```systemverilog
module M0 #(parameter K = 2, parameter type T = logic [K-1:0]) ();
endmodule

module M1;
        M0 #(.K(3)) inst ();
endmodule
```

Output:
```verilog
module M0_CC618;
	parameter type T_K_type;
	parameter T_K_type T_K = 0;
	parameter K = 2;
endmodule
module M1;
	M0_CC618 #(
	        .T_K_type(type(K)),
	        .T_K(K),
	        .K(3)
	) inst();
endmodule
```

Maybe related to #106.

zachjs, 2020-07-10:
Thank you for your continued patience! The issue with default type parameters should be fixed in 737c66a6c99c00c286e95773173a86fb6339edfb.

closed 2020-07-12 by b1f6c1c4