zachjs/sv2v #91: SV Interface support

silkin-limited, 2020-06-03:
I just tried your tool using this simple example: http://www.asic-world.com/systemverilog/interface2.html

//+++++++++++++++++++++++++++++++++++++++++++++++++
// Define the interface
//+++++++++++++++++++++++++++++++++++++++++++++++++
interface mem_if (
  input  wire         clk,
  input  wire         reset,
  input  wire         we,
  input  wire         ce,
  input  wire   [7:0] datai,
  output logic  [7:0] datao,
  input  wire   [7:0] addr
);
endinterface

//+++++++++++++++++++++++++++++++++++++++++++++++++
//   DUT With interface
//+++++++++++++++++++++++++++++++++++++++++++++++++
module simple_if (mem_if mif);
// Memory array
logic [7:0] mem [0:255];
//=================================================
// Read logic
//=================================================
always @ (posedge mif.clk)
 if (mif.reset) mif.datao <= 0;
 else if (mif.ce &&  ! mif.we) begin
   mif.datao <= mem[mif.addr];
 end
//=================================================
// Write Logic
//=================================================
always @ (posedge mif.clk)
 if (mif.ce && mif.we) begin
   mem[mif.addr] <= mif.datai;
 end
endmodule

When I pass the code above as input, I am getting this as output, which is not a valid Verilog code:
module simple_if (mif);
	input mem_if mif;
	reg [7:0] mem [0:255];
	always @(posedge mif.clk)
		if (mif.reset)
			mif.datao <= 0;
		else if (mif.ce && !mif.we)
			mif.datao <= mem[mif.addr];
	always @(posedge mif.clk)
		if (mif.ce && mif.we)
			mem[mif.addr] <= mif.datai;
endmodule

Let me know if this is intended or if I am missing something.

zachjs, 2020-06-04:
Thank you for reporting this! I was unaware of these simple "bundle" interfaces in the specification. I have added support for them on master.

The output for your example is now:
```Verilog
module simple_if (
	mif_clk,
	mif_reset,
	mif_we,
	mif_ce,
	mif_datai,
	mif_datao,
	mif_addr
);
	input wire mif_clk;
	input wire mif_reset;
	input wire mif_we;
	input wire mif_ce;
	input wire [7:0] mif_datai;
	output reg [7:0] mif_datao;
	input wire [7:0] mif_addr;
	reg [7:0] mem [0:255];
	always @(posedge mif_clk)
		if (mif_reset)
			mif_datao <= 0;
		else if (mif_ce && !mif_we)
			mif_datao <= mem[mif_addr];
	always @(posedge mif_clk)
		if (mif_ce && mif_we)
			mem[mif_addr] <= mif_datai;
endmodule
```

Please let me know if you run into any further issues!

silkin-limited, 2020-06-04:
Thanks, your patch works!

closed 2020-06-04 by silkin-limited

poena, 2022-07-23:
> Thank you for reporting this! I was unaware of these simple "bundle" interfaces in the specification. I have added support for them on master.
>
> The output for your example is now:
>
> ```verilog
> module [...]

How to get this result? I've run the same case, and got : 
"Warning: Source includes an interface but output is empty because there is no top-level module which has no ports which are interfaces."

zachjs, 2022-07-23:
@poena The interface conversion has changed substantially in the past two years. Interfaces are now inlined into modules that instantiate them. This means you need to pass sv2v both the interfaces and the modules that use them. Can you give that a try?

poena, 2022-07-26:
Inline is a great feature. But, unroll the interface bundle is also useful. So, can you provide some option to select the coding style?

zachjs, 2022-08-17:
@poena Supporting both styles would not be trivial, though it is theoretically possible. Can you help me understand your specific use case which requires the alternate style?