zachjs/sv2v #93: Conversion of '0 and '1

NilsGraf, 2020-06-14:
`'0` and `'1` are currently converted to `1'sb0` and `1'sb1`, resp. The latter is not accurate for multi-bit assignments (because `'1` actually means that all bits are set to 1, not just the LSB). And the former causes LEC issues when `1'sb0` drives a multi-bit input of a module (resulting in the upper input-bits being undriven).

Would it make sense to use the following translations instead?
-  `1'sb0` to `'d0` (this works fine in LEC for multi-bit inputs)
- `1'sb1` to `-'sd1` (note that signed -1 is all-ones)

See also [this workaround](https://github.com/lowRISC/ibex/pull/950/files#diff-091b97af7102221f1b5b9af1b2561045R51-R56).  Thanks!

zachjs, 2020-06-15:
Handling these "unbased unsized" literals has proven deceptively complex. In self-determined contexts (i.e., `{'1, 'x}`), they should behave as though they are a single bit. Otherwise, they fill the context-supplied size. 

As far as I can tell `'d0` is exactly the same as `0`, and `-'sd1` is exactly equivalent to `-1` (but not `-'sb1`). I recently added a hack to deal with the sizing associated with ternary expressions. Your suggestion of `-'sd1` thankfully eliminates the need for that hack. However, I think to make this conversion work properly, it must be context aware, selecting either the sized or unsized replacement depending on where it is used..

NilsGraf, 2020-06-15:
Awesome. I think mapping `'0` and `'1` onto `0` and `-1` should also work fine.

zachjs, 2020-06-15:
I think 8cfd05de1a821cc91986643b48249e637882b81f should be closer to what you're looking for. Please let me know if you find any issues.

zachjs, 2020-06-19:
Has this issue been resolved?

NilsGraf, 2020-06-19:
Yes, thanks!

closed 2020-06-19 by zachjs