zachjs/sv2v #94: Translation of 'parameter bit'

NilsGraf, 2020-06-14:
Example from [here](https://github.com/lowRISC/opentitan/blob/master/hw/ip/prim/rtl/prim_flop_2sync.sv#L9
): 
```systemverilog
parameter bit ResetValue = 0
```
Gets translated to
```systemverilog
  parameter ResetValue = 0;
```
The generated parameter is multi-bit by default. Which then causes issues in [line 21](https://github.com/lowRISC/opentitan/blob/master/hw/ip/prim/rtl/prim_flop_2sync.sv#L21) in this `{Width{ResetValue}}`.

It works when I explicitly specify the bitwidth of the parameter as follows:
```systemverilog
  parameter [0:0] ResetValue = 0;
````
Thanks!

zachjs, 2020-06-14:
Thank you for finding and reporting this issue! Single bit parameters are now explicitly sized as suggested.

closed 2020-06-14 by zachjs

NilsGraf, 2020-06-15:
Awesome! Thanks for the prompt response.

Quick question: Will it also work correctly for bit-vectors such as below (see also [here](https://github.com/lowRISC/opentitan/blob/master/hw/ip/tlul/rtl/tlul_socket_1n.sv#L45))?
```systemverilog
parameter bit [3:0]     HReqDepth = 4'h2,
```

zachjs, 2020-06-15:
Yes, that is still handled correctly. Under the hood, `parameter bit` becomes `parameter logic`, and then the `logic` is elaborated away.

While
`parameter bit [3:0]` => `parameter logic [3:0]` => `parameter [3:0]`
works as expected,
`parameter bit` => `parameter logic` => `parameter`
clearly does not.

The conversion responsible for that second step no longer loses the single-bit sizing. There is no change in behavior for the other case.

NilsGraf, 2020-06-15:
Awesome, thanks for clarifying.

commit cbe0071e (2020-06-14): fix bit param sizing (resolves #94)