zachjs/sv2v #95: Parameter value from interface missing in generated output

tpoikela, 2020-06-14:
Hi. I tried the sv2v v0.0.3 (b71e0f5) with this input:

```
interface test_if();

parameter int DW = 32;

logic[DW-1:0] data;

modport consumer (
    input data
);

endinterface

module test_mod(
    input clk,
    input rst,
    test_if.consumer c_if,

    output[31:0] dout
);

reg[31:0] reg_ff;
assign dout = reg_ff;

always_ff@(posedge clk or negedge rst)
if (rst == 1'b0) begin
    reg_ff <= 0;
end
else begin
    reg_ff <= c_if.data;
end

endmodule
```

Expected result: `parameter DW = 32` is included in the output.

Actual result: `parameter DW` value is missing. This is the output: 

```
module test_mod (
        clk,
        rst,
        c_if_data,
        dout
);
        input clk;
        input rst;
        input wire [DW - 1:0] c_if_data;
        output [31:0] dout;
        reg [31:0] reg_ff;
        assign dout = reg_ff;
        always @(posedge clk or negedge rst)
                if (rst == 1'b0)
                        reg_ff <= 0;
                else
                        reg_ff <= c_if_data;
endmodule
```

zachjs, 2020-06-15:
Thank you taking the time to file this issue! It has actually called attention to a couple of issues. sv2v was not correctly resolving the parameters of an interface instantiated with positional parameter bindings. Modules containing modport ports lost the interface's parameters, as you mentioned. Instances of those modules, too, were missing the appropriate parameter bindings. These should all be fixed as of 99428b2f16ed4317e1a5212663c551fb47570f9c.

tpoikela, 2020-06-15:
Thanks! It works now!

closed 2020-06-15 by tpoikela