zachjs/sv2v #98: For loops in SystemVerilog not translating

nagendragd, 2020-06-22:
I am using the IBEX project (https://github.com/lowRISC/ibex) and its synthesis flow uses sv2v to convert everything to Verilog before invoking synthesis on the design. 

When I run the standard flow, I see that sv2v reports errors while processing some for-loops.

I tried both the precompiled sv2v installation as well as building my own from GIT. Both reported the same error. 

I have attached one of the IBEX SV files (renamed to .txt since github did not allow me to attach the file with .sv extension) that shows an error in line number 326 corresponding to the for loop statement:

 for (int unsigned i=0; i<32; i++) begin
      shift_result_rev[i] = shift_result[31-i];
 end

Best Regards
Nagendra

[ibex_alu.txt](https://github.com/zachjs/sv2v/files/4816095/ibex_alu.txt)

zachjs, 2020-06-22:
I can't reproduce your issue. Could you share more information? I'm using
- `sv2v v0.0.4 (0fb97f2)`
- https://github.com/lowRISC/ibex/commit/ae547c8d3010d86ef7afdee900d461616c3cb414

nagendragd, 2020-06-23:
Well, here's the message I get when I run sv2v in verbose mode on that file:

sv2v: invalid for loop decl: // Trace: ../rtl/ibex_alu.sv:326:14
CallStack (from HasCallStack):
  error, called at src/Convert/ForDecl.hs:55:5 in main:Convert.ForDecl

And about sv2v version that I am on:
ndg0068@alveo-cse:~/rv_design/ibex/syn$ sv2v --help
sv2v v0.0.4 (0fb97f2)

See any issues?

Thanks for your time!

Best Regards
Nagendra

zachjs, 2020-06-23:
Thanks for spotting this! It looks like you're using sv2v's verbose mode, and indeed found a bug when using that mode with for loops containing a declaration in the instantiation. I have fixed this issue, and added coverage for verbose mode to the test runner as of 2d134a864000c6a4163db64a115245201bf06a41.

nagendragd, 2020-06-23:
That fix did the trick! Thanks!!

closed 2020-06-23 by nagendragd