zachjs/sv2v #101: package file not getting converted

nagendragd, 2020-06-25:
I am not completely sure that I have it right, but wanted to report it to see if there's a usage issue or an sv2v issue. The attached package file is not converting - sv2v produces an empty output. Verbose mode does not print any warnings or messages. I am not sure what the expected behavior is though. I have attached the .sv source file along with for your reference (please rename .txt to .sv). I am on the latest sv2v tool version.

Best Regards
Nagendra
[el2_defines.txt](https://github.com/zachjs/sv2v/files/4834073/el2_defines.txt)

zachjs, 2020-06-26:
Higher level constructs like packages, interfaces, and typedefs cannot be represented in Verilog. The conversion procedures for these constructs typically involves making them and their underlying definitions available to the rest of the input source, and performing the appropriate substitutions, and then removing the high level constructs so the output can be consumed by downstream tools.

nagendragd, 2020-06-26:
I get it. However, I have a follow up question: the Verilog files whose SV files had the import statements for the package continue to have the same import statement. Do imports work at all with Verilog? I thought not. For your reference, I have attached the SV file with the import and the corresponding Verilog file. File extensions renamed to .txt to keep github happy..  el2_swerv_sv.txt is the SV source and el2_swerv_v.txt is the Verilog output. Does the Verilog output look correct to you?

[el2_swerv_sv.txt](https://github.com/zachjs/sv2v/files/4835013/el2_swerv_sv.txt)
[el2_swerv_v.txt](https://github.com/zachjs/sv2v/files/4835014/el2_swerv_v.txt)

zachjs, 2020-06-26:
sv2v can only perform conversions based on the files it is given. To convert an `import`, for example, the definition of the package being imported must be available. In this case, sv2v should be invoked with both `el2_defines.sv` and `el2_swerv.sv` simultaneously. I confirmed that with both of those files passed in (although I am missing `el2_params.svh`), the `import` is indeed elaborated away.

nagendragd, 2020-06-26:
Thanks. That worked! It inserted definitions from the package file into files which imported them.

However, and this may be another usage issue with how I am invoking sv2v, I now encounter a parameter definition related syntax error in the generated verilog (I am running this verilog through Yosys which reports the error).

Assuming that all the files are in the same folder, my invocation is:
sv2v --verbose ./el2_def.sv -I. el2_swerv.sv > el2_swerv.v

To help reproduce the issue,  I have attached el2_params.vh file that houses the parameter definition. In the generated verilog, line number 322, it has the elaborated parameter definition which Yosis and Verilator complain about. For eg: Verilator shows:
 verilator --lint-only el2_swerv.v
%Error: el2_swerv.v:322:24: syntax error, unexpected IDENTIFIER, expecting ';'
  322 |  parameter el2_param_t pt = '{
      |                        ^~
%Error: Exiting due to 1 error(s)

(I have attached both the SV sources of el2_swerv, el2_def and el2_params, and the generated el2_swerv verilog -- all suitably renamed )

Do you think my sv2v invocation is incorrect? Thanks much!

[el2_def_sv.txt](https://github.com/zachjs/sv2v/files/4837695/el2_def_sv.txt)
[el2_param_vh.txt](https://github.com/zachjs/sv2v/files/4837697/el2_param_vh.txt)
[el2_swerv_sv.txt](https://github.com/zachjs/sv2v/files/4837698/el2_swerv_sv.txt)
[el2_swerv_v.txt](https://github.com/zachjs/sv2v/files/4837699/el2_swerv_v.txt)

zachjs, 2020-06-26:
Thank you for sharing those files. There are at least two outstanding issues:
1) `el2_param_t` is not defined in any of these files
2) The source uses `.*` for wildcard port bindings. These are not allowed in Verilog, but the only way sv2v can convert them is to also be given the definition for those modules so it knows what ports they have.

nagendragd, 2020-06-26:
Oh right. I will sort out issue 1. first. Thank you again. If you are ok, I will leave this issue thread open for now so that if there's something I run into that I want to share, I don't have to open a new ticket.

nagendragd, 2020-06-26:
I think I sorted out #1. My new invocation is:
sv2v --verbose ./el2_def.sv ./el2_pdef.vh -I. el2_swerv.sv > el2_swerv.v

(the new file el2_pdef.vh has the parameter type definition)

But I ran into some new error that I will share separately ..

nagendragd, 2020-06-26:
After I fixed the invocation, sv2v generates verilog with the parameter elaborated. 
However, yosys encounters and stops with the below error:

Executing Verilog-2005 frontend: syn_out/swerv_el2_26_06_2020_12_17_13/generated/el2_dec_decode_ctl.v
syn_out/swerv_el2_26_06_2020_12_17_13/generated/el2_dec_decode_ctl.v:1013: ERROR: 2nd expression of generate for-loop is not constant!

Is this also due to my incorrect invocation of sv2v?

I have attached the relevant files.. 
[el2_dec_decode_ctl_sv.txt](https://github.com/zachjs/sv2v/files/4838345/el2_dec_decode_ctl_sv.txt)
[el2_dec_decode_ctl_v.txt](https://github.com/zachjs/sv2v/files/4838346/el2_dec_decode_ctl_v.txt)
[el2_def_sv.txt](https://github.com/zachjs/sv2v/files/4838347/el2_def_sv.txt)
[el2_param_vh.txt](https://github.com/zachjs/sv2v/files/4838348/el2_param_vh.txt)
[el2_pdef_vh.txt](https://github.com/zachjs/sv2v/files/4838349/el2_pdef_vh.txt)

zachjs, 2020-06-26:
The result you've attached still contains `.*`. However, based on your error message, the issue you're encountering is likely a Yosys bug similar to the one I fixed in https://github.com/YosysHQ/yosys/pull/2136. sv2v has generated a loop containing a constant function call, which is perfectly legal, but Yosys is not properly elaborating it. I will probably take a look at fixing that issue soon.

nagendragd, 2020-06-26:
Oh ok thanks! Yes, I still need to sort out the .*.
Are you also developing/maintaining Yosys? Very cool! 

I look forward to the Yosys fix..

zachjs, 2020-06-26:
I don’t maintain Yosys, but I have made a couple small contributions to its Verilog frontend. Certain sv2v conversions have yielded valid Verilog that Yosys didn’t support. Sometimes making changes to Yosys is easier (or more principled) than changing sv2v’s behavior or asking end users to employ some kind of workaround.

nagendragd, 2020-06-26:
Yes it makes sense fixing Yosys where the error really is rather than working around it. I am still trying to get a measure of how to get the .* bindings out of the Verilog code. By exposing all the module definitions to sv2v, I see that it inlined those referenced modules and I now see module re-definition errors reported by Yosys. Further, some modules instantiate several other modules and it seems my command line has to supply all of these referenced modules. Am I doing this correctly?  

What I am doing:
sv2v <referenced module 1.sv> <referenced module 2.sv> .. <referenced module N.sv> <module.sv>

zachjs, 2020-06-26:
sv2v converts all of files it is given. The earlier files are not just used for resolution. Because of the cross dependence of the source files, it probably is not feasible to convert each file/module individually. My suggestion would be to pass all of your source files (but not included headers) to sv2v all at once, and take the resulting (probably large) file and pass that into Yosys.

nagendragd, 2020-06-26:
Ok that was also a question I had if the other files (all except the last one) were used only for resolution or not.  Your response sorted me out.  I assume that the ordering of input files does not matter and that sv2v will resolve all dependencies regardless of file order.

Yes I started with the IBEX set up where files were converted individually. There are may be about 30 *.sv files (not including headers) that I will specify all at once (with the -I for the include paths to pick up headers) and see how it all works out.

nagendragd, 2020-06-26:
Ok - I tried this suggestion and after having to include one header file explicitly, I generated the attached one single verilog output. It has resolved the parameter and module port bindings. I don't see any .*'s any more.

I ran it through verilator --lint-only and see it report several warnings. I have attached that log as well if there's anything that concerns sv2v that you want to take a look at. But I do not see any errors reported by lint which I take as good news.

If this output looks good to you, then I just have to wait for the Yosys fix. 
Please let me know if you find any issues related to invoking sv2v or inside sv2v itself.

Thanks once again for your patience with me..
[veri.log.txt](https://github.com/zachjs/sv2v/files/4839461/veri.log.txt)
[el2_swerv.v.txt](https://github.com/zachjs/sv2v/files/4839462/el2_swerv.v.txt)

zachjs, 2020-06-27:
I have updated that pending Yosys PR to allow constant function calls in the condition of for loops. This helps the Yosys frontend progress further, but it looks it's encountering some issues with having `genvar i` in the same generate block as a procedural loop using a different `i`.

I haven't had the chance to look at all of the warnings, but at least some of them are expected.

nagendragd, 2020-06-27:
Thanks for the update. I will wait for your confirmation of the full fix in
Yosys (or a workaround if you think that’s the easier path for now).

Have a great weekend!

zachjs, 2020-07-01:
The Yosys PR related to constant functions has been merged. I believe the key remaining issue in processing the output you're getting is described in https://github.com/YosysHQ/yosys/issues/2214. I took a stab at fixing these myself, but I think it will take someone more familiar with the codebase to fix it properly.

nagendragd, 2020-07-01:
> YosysHQ/yosys#2214 <https://github.com/YosysHQ/yosys/issues/2214>. I took

Thanks for the update. Should I try anything in the mean time? And thanks
for filing the Yosys issue..

Best regards
Nagendra

zachjs, 2020-07-01:
I would suggest trying to split up the block `ECC1_MUX`.

zachjs, 2020-07-09:
@nagendragd Is there still an outstanding issue with sv2v, or can this be closed?

nagendragd, 2020-07-09:
I think we will close this issue on sv2v. Hopefully the Yosys developers will get the proper Yosys fix. I believe you have already filed an issue over there, right?

zachjs, 2020-07-09:
Yes, the issue is linked above.

nagendragd, 2020-07-09:
Yes I am marking this closed.. thank you for your patience and help. Much appreciated..

closed 2020-07-09 by nagendragd