zachjs/sv2v #99: Bug when accessing a field of a function's return value

b1f6c1c4, 2020-06-25:
Input:
```systemverilog
package PKG;
  typedef struct packed {
    logic F;
  } S;
  function automatic S f();
    return '0;
  endfunction
endpackage

module M;
  localparam L = PKG::f().F;
endmodule
```

Output:
```verilog
module M;
	function automatic [0:0] PKG_f;
	        input _sv2v_unused;
	        PKG_f = 'sd0;
	endfunction
	localparam L = PKG_f(0)[0]; // syntax error
endmodule
```

To my understanding verilog does NOT allow `[]` to follow any expression other than base expressions.

zachjs, 2020-06-25:
You are correct that using bit or range selects directly on the result of a function call is forbidden. However, accessing struct members in this context is also forbidden. For reference, I am looking at the definitions of `primary` and `select` in Appendix A.8.4 (page 1174) of the SystemVerilog-2017 specification.

Because of this, I believe the example provided is illegal. I’m not a huge fan of this aspect of the specification, and I wouldn’t be surprised in future versions of SystemVerilog loosened these restrictions.

zachjs, 2020-06-25:
I think unfortunately this example is still invalid. I believe `L0` needs to be explicitly declared to have type `PKG::S`. Commercial tools also appear to impose this restriction.

b1f6c1c4, 2020-06-25:
@zachjs I've updated the test case again. Adding `PKG::S` won't help. Could you please double check?

zachjs, 2020-06-26:
Nice catch! The `$clog2` is a target for sv2v's general elaboration/simplification conversion. As of 82d06b39155654f4d6d8c97aa266346250e58d67, this conversion should no longer perform substitutions which result in illegal `select`s.

closed 2020-06-26 by b1f6c1c4