zachjs/sv2v #102: Issue with unsized '1 in comparisons

nmoroze, 2020-06-30:
Hi, I think I've found an issue related to #93 - it has to do with unexpected behavior using `'1` in a comparison. 

Here's an example SV snippet:
```systemverilog
module top (
);

logic [3:0] a;
logic b;

initial begin
    a = 4'hf;
    b = (a != '1);
    $display(b);
end

endmodule
```

Which converts to the following using sv2v:
```verilog
module top;
	reg [3:0] a;
	reg b;
	initial begin
		a = 4'hf;
		b = a != -'sd1;
		$display(b);
	end
endmodule
```

Since the behavior of `'1` is to fill with 1's to a value's context-dependent width, I'd expect `b` to equal 0. However, running it under Icarus Verilog, the value 1 is printed. 

Synthesizing a similar circuit in Yosys gives a bit more insight into the behavior. 
```verilog
module top (
	in,
	out
);
	input [3:0] in;
	output out;
	assign out = in != -'sd1;
endmodule
```

Here, synthesis appears to generate a circuit where the `-'sd1` is sign-extended to fill 32 bits, while `in` is zero-extended to fill 32 bits (so the two are never equal). 

It seems to me like these tools do not consider the implicit width of these comparisons to be equal to the largest operand (whereas perhaps SystemVerilog would for the case of a `'1`?) I'm not entirely sure if this is to Verilog spec, but it seems like the `-sd1` replacement for `'1` does not work soundly in the case of comparisons. 

Thanks for taking a look!

NilsGraf, 2020-06-30:
Just curious, would `(~ 'd0)` work?

nmoroze, 2020-06-30:
Interesting thought @NilsGraf, that seems to have the seem behavior though.  

By the way, I noticed that both you and @zachjs seem to be experimenting with using a commercial LEC tool to verify that the sv2v translation of the LowRISC Ibex is correct. This case actually came from the OpenTitan, in particular this line: https://github.com/lowRISC/opentitan/blob/master/hw/ip/tlul/rtl/tlul_adapter_sram.sv#L188 (the `tl_i.a_mask != '1` being the isolated problematic snippet). 

Are you planning/have you looked at using the LEC tool to verify the OpenTitan translation as well?

NilsGraf, 2020-07-01:
Yes, we are actually seeing the same issue in LEC.

It's interesting that the ~ 'd0 also fails. Is the following correct?
```systemverilog
(4'b0000 == 'd0)       // evaluates to 1.
(4'b1111 == (~ 'd0))   // evaluates to 0?  Or 1?
(4'b1111 == 5'b11111)  // evaluates to 0?  Or 1?
```

nmoroze, 2020-07-01:
Ran these in Icarus Verilog and got the following:
```verilog
module test();

initial begin
    $display(4'b0000 == 'd0);    // => 1
    $display(4'b1111 == (~ 'd0));   // => 0
    $display(4'b1111 == 5'b11111);  // => 0
    $display(~ 4'b1111 == 'd0);     // => 0
end

endmodule
```

The last result in particular seems a bit odd to me...

NilsGraf, 2020-07-01:
Hi @zachjs, it's unfortunate, but it seems sv2v would need to figure out the bitwidth of the left-hand side and then replace `'1` by `{ bit_width_of_lhs { 1'b1 }}`?

zachjs, 2020-07-01:
I'd like to get a sense for the full set of changes necessary before getting started. Do we know if similar sizing issues exist with the current conversion strategy when unbased unsized literals are used in any of the following contexts?
- port bindings (I believe there are instances of this in OpenTitan)
- task/function arguments
- binary operators other than comparisons (i.e. `4'b1111 ^ '1`)

NilsGraf, 2020-07-01:
It seems OpenTitan only needs the comparison case (for `tl_i.a_mask != '1` in [line 188 of tlul_adapter_sram.sv](https://github.com/lowRISC/opentitan/blob/master/hw/ip/tlul/rtl/tlul_adapter_sram.sv#L188). Alternatively, I wonder if Yosys already supports `'1`. In this case, sv2v could leave `'1` untouched (but this wouldn't work for other tools such as Icarus).

nmoroze, 2020-07-01:
There's at least one of other instance I can think of for the OpenTitan in particular, which is [line 128 in spi_fwm_rxf_ctrl.sv](https://github.com/lowRISC/opentitan/blob/master/hw/ip/spi_device/rtl/spi_fwm_rxf_ctrl.sv#L128). It's an assignment though, which we already handle fine.

> Alternatively, I wonder if Yosys already supports '1. In this case, sv2v could leave '1 untouched (but this wouldn't work for other tools such as Icarus).

This actually does seem to be the case! In fact, it seems that the `-g2005-sv` flag makes Icarus handle `'1` on its own as well. Perhaps this is something it would make sense to pass through?

If you still want to try and convert it in sv2v though, I'd be happy to write test cases for the scenarios you described @zachjs.

nmoroze, 2020-07-01:
I've also noticed another issue related to unsized integer literals that comes from the Ibex (although an older version than current top-of-tree). 

[Line 55 of ibex_register_file_ff.sv](https://github.com/lowRISC/ibex/blob/d14312c3cc47a7d4b34943e4febb96312353b2cc/rtl/ibex_register_file_ff.sv#L55) in d14312 has the line `rf_reg_tmp <= '{default:'0};`, which gets converted to `rf_reg_tmp <= {(< some constant expr >) {'sd0}};`. This seems to be handled okay by Yosys, but once again Icarus gets unhappy here, throwing the error "Concatenation operand "'sd0" has indefinite width."

Not sure how related you'd consider this one, happy to open another issue about it if you'd prefer!

zachjs, 2020-07-02:
I don’t think Yosys supports these literals. I believe that mode in iverilog is referring to the original version of the SystemVerilog specification. I would definitely like to have sv2v have full support, so any test cases are greatly appreciated.

zachjs, 2020-07-07:
95208947204bf19ee524afa2e8ee5f78eee9896f contains some major improvements to the conversion for these literals. In general, they are given explicit appropriate sizing in a context-aware fashion. I think the `'sd` alternatives were a bit of a red herring in that they don't automatically size, but are rather implicitly 32 bits, which happened to be larger than the ports for which they were being used.

zachjs, 2020-07-09:
@nmoroze @NilsGraf Have you had the chance to test out the new behavior? I'm pretty hopeful that we should _finally_ be done dealing with these literals.

nmoroze, 2020-07-09:
Just tried it out, everything that was broken for me looks to be fixed now. Thanks so much! 

Will close for now, but curious to hear from @NilsGraf if this looks cleaned up in the LEC as well (if it's not a lot of hassle to check)?

closed 2020-07-09 by nmoroze