zachjs/sv2v #103: Duplicate struct conversion functions generated

nmoroze, 2020-07-01:
I've noticed an issue where in some particular circumstance sv2v emits duplicates of its autogenerated struct-creation functions.

Unfortunately I had a hard time finding a minimal example for this, but I've linked the file that demonstrates the problem (and a dependency). It comes from OpenTitan 97b60c11 (the problem doesn't appear to exist on the current top-of-tree, not sure why). 

Files:
https://raw.githubusercontent.com/lowRISC/opentitan/97b60c11ab4cbc42b579d3ebc61d43c69843efb1/hw/ip/rv_core_ibex/rtl/rv_core_ibex.sv
https://raw.githubusercontent.com/lowRISC/opentitan/97b60c11ab4cbc42b579d3ebc61d43c69843efb1/hw/ip/tlul/rtl/tlul_pkg.sv

If you download these and run sv2v on both, the following functions are duplicated at the bottom of the output: `sv2v_struct_8D9F8` and `sv2v_struct_50735`.

This doesn't seem to be a problem for Yosys, but it does cause Icarus to throw a duplicate definition for function error.

zachjs, 2020-07-02:
I've wanted to remove these "stuct packers" for some time, as they basically amount to grouped size casts. The struct conversion predates the size cast conversion. I had already changed sv2v to replace struct literals with a concatenation of explicit casts on a separate branch. This change has been pushed as 1dfa9a9e7ffeb6dddf33b808d8e8dfa3a1411830.

nmoroze, 2020-07-06:
Thanks, that change fixes my problem!

closed 2020-07-06 by nmoroze