zachjs/sv2v #105: Improper handling of wide parameters

b1f6c1c4, 2020-07-09:
Input:
```systemverilog
module M;
	localparam [63:0] X = 1 << 63;
endmodule
```
Output:
```systemverilog
module M;
	localparam [63:0] X = 9223372036854775808;
endmodule
```
The `9223372036854775808` here is NOT a valid int.

Actually I'm wondering if `1 << 63` is actually legal or not. I'm not super familiar with Verilog standard, so maybe you can better answer this question for me.

zachjs, 2020-07-09:
`1 << 63` is perfectly legal. In fact, as far as I can tell, `9223372036854775808` is also legal.

From Section 5.7.1:
> The number of bits that make up an unsized number (which is a simple decimal number or a number without the size specification) shall be at least 32.

Am I missing something here? Is there a particular downstream tool which has trouble with these constants?

b1f6c1c4, 2020-07-09:
I'm using Verilator.

Besides, in my opinion, "shall be at least 32." does not guarantee that `9223372036854775808` is legal; instead, it means that any number less than or equals to `4294967295` is legal.

zachjs, 2020-07-09:
I've updated master to avoid generating such literals during constant folding. Your example now yields:
```Verilog
module M;
	localparam [63:0] X = 64'd9223372036854775808;
endmodule
```

b1f6c1c4, 2020-07-12:
It looks that this becomes a problem again after d88c516d33c7aef93e3f6b594596adc513329c07. `sv2v v0.0.4 (d88c516)` is producing `'sd9223372036854775808`.

zachjs, 2020-07-13:
This is fixed again in b2291a204634d3fae2b76e6fe7f7481598b98a35. `iverilog` doesn't have trouble with constants such as `sd9223372036854775808`, so having test coverage under the existing setup isn't straightforward.

closed 2020-07-13 by b1f6c1c4