zachjs/sv2v #111: sv2v: src/Convert/TypeOf.hs:181:11-37: Non-exhaustive patterns in (tf, _ : rs)

stephenry, 2020-07-19:
Running latest SV2V I see the following error message:

```shell
sv2v: src/Convert/TypeOf.hs:181:11-37: Non-exhaustive patterns in (tf, _ : rs)
```

Failing code appears to be the "if (dealloc_spec_vld)..." and "if (dealloc_vld)..." statements in the code below:

```systemverilog
  logic
    [pa_engine_pkg::PHemiN - 1:0]
    [pa_engine_pkg::PChN - 1:0]
    [1:0]                                    flm_dealloc_vld;
  pa_pkg::thread_t
    [pa_engine_pkg::PHemiN - 1:0]
    [pa_engine_pkg::PChN - 1:0]
    [1:0]                                    flm_dealloc_1h;

  always_comb begin : dealloc_cntrl_PROC

    // Defaults:
    flm_dealloc_vld  = '0;
    flm_dealloc_1h   = '0;

    // For each hemisphere
    for (int h_id = 0; h_id < pa_engine_pkg::PHemiN; h_id++) begin
      // For each channel
      for (int ch_id = 0; ch_id < pa_engine_pkg::PChN; ch_id++) begin
        if (dealloc_spec_vld)
          flm_dealloc_vld [h_id][ch_id][0]  = (dealloc_spec [h_id][ch_id] != '0);
        if (dealloc_vld)
          flm_dealloc_vld [h_id][ch_id][1]  = (dealloc [h_id][ch_id] != '0);
      end
    end

    // For each hemisphere
    for (int h_id = 0; h_id < pa_engine_pkg::PHemiN; h_id++) begin
      // For each channel
      for (int ch_id = 0; ch_id < pa_engine_pkg::PChN; ch_id++) begin
        flm_dealloc_1h [h_id][ch_id][0]  = dealloc_spec [h_id][ch_id];
        flm_dealloc_1h [h_id][ch_id][1]  = dealloc [h_id][ch_id];
      end
    end

  end // block: dealloc_cntrl_PROC

```

zachjs, 2020-07-19:
Thank you for taking the time to report this issue! I don't think I can properly track it down without seeing the declarations for `dealloc_spec` and `dealloc`. Could you share them with me?

stephenry, 2020-07-20:
Hi Zach,

Thanks for your help. The following self-contained example reproduces the problem:

[fail.sv.gz](https://github.com/zachjs/sv2v/files/4950017/fail.sv.gz)

zachjs, 2020-07-20:
Thank you for sharing that with me. The conversion responsible for determining the types of certain expressions did not appropriately handle bit/range select expressions whose types were aliases that had not yet been fully resolved. Your given example now converts successfully, save for resolving the type of `thread_t`, which appears to be missing.

As a side note, sv2v supports and validates closing tags on procedural blocks, so you could replace:
```SystemVerilog
end // block: dealloc_cntrl_PROC
```
with:
```SystemVerilog
end : dealloc_cntrl_PROC
```

closed 2020-07-20 by zachjs

stephenry, 2020-07-21:
That did the trick! Thank you for your help!

commit e88a6b9d (2020-07-20): fix type lookup of yet-unresolved types (resolves #111)