zachjs/sv2v #113: Bug when using '{} on multi-dimension array

b1f6c1c4, 2020-07-20:
I don't know if this is valid SystemVerilog though, but here is the code snippet:
```systemverilog
module M;
  typedef struct packed {
    logic [1:0] f[0:1];
  } T;
  localparam T p = '{f:'{'{default:'d1},'{default:'d2}}};
endmodule
```
Output from `sv2v v0.0.4 (e6e62e8)`:
```verilog
module M;
        function automatic [3:0] sv2v_cast_4;
                input reg [3:0] inp;
                sv2v_cast_4 = inp;
        endfunction
        localparam [3:0] p = sv2v_cast_4({'{
                default: 'd1
        }, '{
                default: 'd2
        }});
endmodule
```
It contains non-verilog syntax.

zachjs, 2020-07-20:
Most tools appear to dislike the use unpacked dimensions within `struct packed`, though sv2v handles this just fine. Your given example now converts successfully, although the implicit cast of `'d2` to `1'b0` may be a bit surprising.

closed 2020-07-20 by zachjs

b1f6c1c4, 2020-07-21:
I tried something similar with #112 as below:
```systemverilog
package PKG;
  typedef struct packed {
    logic f;
  } foo_t;
endpackage
module M;
  typedef struct packed {
    PKG::foo_t f;
  } local_t;
  local_t w[0:1];
  always_comb begin
    w <= '{default: 0};
  end
endmodule
```
And the output from `sv2v v0.0.4 (49c0d29)` is:
```verilog
module M;
        reg [1:0] w;
        always @(*) w <= {2 {{2 {{2 {'{
                f: 0
        }}}}}}};
endmodule
```
So I believe the fix is NOT complete in this issue.

zachjs, 2020-07-22:
I've taken another stab at the handling of these literals. This should be fixed as of 5ad8de9ef7a2aec8da86a288f1796d2f28602d1e.

b1f6c1c4, 2020-07-22:
@zachjs Thanks you so much for fixing all (25) the issues! After this fix I am finally able to push the entire[Ariane/CVA6](https://github.com/openhwgroup/cva6) through sv2v and build with VCS (and Verilator maybe) :tada:. However the output Verilog from sv2v cannot pass the tests. Should I confirm more bugs in the future, I'll submit them to you.

zachjs, 2020-07-25:
I spotted and fixed a couple issues in the generated output for Ariane. I'm hoping to try to put it through formal verification in the near future.

commit 7eed2fc5 (2020-07-20): fix handling of multi-dimensional integer array literals (resolves #113)