zachjs/sv2v #114: Interface members declared inout in toplevel module.

Zottel, 2020-07-20:
In my tests, an interface declared in my top module to connect two other modules turned
into inout wires instead of plain wires.

The following:
```SystemVerilog
interface transferbytes;
        logic req;
        logic ack;
        logic [7:0] data;
        modport producer ( output req, data,
                           input ack);
        modport consumer ( output ack,
                           input req, data);
endinterface;
module main (clk);
...
        transferbytes itb;
...
...
```
turned into
```Verilog
module main (clk);
...
        inout wire itb_req;
        inout wire itb_ack;
        inout wire [7:0] itb_data;
```
which is wrong since that interface is not a module parameter.

To reproduce, the following SystemVerilog...
```SystemVerilog
module consumer (input clk, transferbytes.consumer int_bytes);
        assign int_bytes.ack = 1;

        always @ (posedge clk) begin
                if (int_bytes.req)
                        $display("consume new value: %b", int_bytes.data);
        end
endmodule;

module producer (input clk, transferbytes.producer int_bytes);
        reg [7:0] state = 0;

        assign int_bytes.data = state;

        always @ (posedge clk) begin
                if (int_bytes.ack) begin
                        if (state != 8'b10101010) begin
                                state <= state + 1;
                                $display("produce new value: %b", (state+1));
                                int_bytes.req <= 1;
                        end else
                                int_bytes.req <= 0;
                end else
                        int_bytes.req <= 0;
        end
endmodule;
interface transferbytes;
        logic req;
        logic ack;
        logic [7:0] data;
        modport producer ( output req, data,
                           input ack);
        modport consumer ( output ack,
                           input req, data);
endinterface;
module main (clk);
        input clk;

        reg [16:0] steps = 0;
        reg first_clock = 1;

        transferbytes itb;

        producer prod (clk, itb);
        consumer cons (clk, itb);

        always @ (posedge clk) begin
                if (first_clock) begin
                        $display("Hello World");
                        first_clock <= 0;
                end else
                        first_clock <= 0;

                if (steps > 1024)
                        $finish;
                else
                        steps <= steps + 1;

        end
endmodule
```
...generates the following Verilog:
```Verilog
module consumer (
        clk,
        int_bytes_ack,
        int_bytes_req,
        int_bytes_data
);
        input clk;
        output wire int_bytes_ack;
        input wire int_bytes_req;
        input wire [7:0] int_bytes_data;
        assign int_bytes_ack = 1;
        always @(posedge clk)
                if (int_bytes_req)
                        $display("consume new value: %b", int_bytes_data);
endmodule
module producer (
        clk,
        int_bytes_req,
        int_bytes_data,
        int_bytes_ack
);
        input clk;
        output reg int_bytes_req;
        output wire [7:0] int_bytes_data;
        input wire int_bytes_ack;
        reg [7:0] state = 0;
        assign int_bytes_data = state;
        always @(posedge clk)
                if (int_bytes_ack) begin
                        if (state != 8'b10101010) begin
                                state <= state + 1;
                                $display("produce new value: %b", state + 1);
                                int_bytes_req <= 1;
                        end
                        else
                                int_bytes_req <= 0;
                end
                else
                        int_bytes_req <= 0;
endmodule
module main (clk);
        input clk;
        reg [16:0] steps = 0;
        reg first_clock = 1;
        inout wire itb_req;
        inout wire itb_ack;
        inout wire [7:0] itb_data;
        producer prod(
                .clk(clk),
                .int_bytes_req(itb_req),
                .int_bytes_ack(itb_ack),
                .int_bytes_data(itb_data)
        );
        consumer cons(
                .clk(clk),
                .int_bytes_req(itb_req),
                .int_bytes_ack(itb_ack),
                .int_bytes_data(itb_data)
        );
        always @(posedge clk) begin
                if (first_clock) begin
                        $display("Hello World");
                        first_clock <= 0;
                end
                else
                        first_clock <= 0;
                if (steps > 1024)
                        $finish;
                else
                        steps <= steps + 1;
        end
endmodule
```

zachjs, 2020-07-20:
Thank you for taking the time to file this issue!

From the SystemVerilog-2017 grammar:
<pre><code>interface_instantiation ::= interface_identifier [ parameter_value_assignment ] hierarchical_instance { <b>,</b> hierarchical_instance } <b>;</b>
hierarchical_instance ::= name_of_instance <b>(</b> [ list_of_port_connections ] <b>)</b>
name_of_instance ::= instance_identifier { unpacked_dimension }
</pre></code>
My interpretation is that an interface instantiation must contain a list of port connections in parentheses, even if it is empty. Adding those parentheses (i.e., `transferbytes itb()`) yields reasonable output from sv2v.

I tried a few different commercial simulators on your example. All of them produce an error due to the missing parentheses. To avoid confusion, I will add an error message to sv2v when this scenario is encountered.

Zottel, 2020-07-21:
I see, makes sense, thank you for the clarification. :)
(That also explained the confusing error in a commercial simulator.)

closed 2020-07-23 by Zottel