zachjs/sv2v #116: Cast function in generate block

stephenry, 2020-07-22:
Hi Zach,

Another little problem. In the code attached, I'm seeing a function being emitted bv SV2V in the scope of a generate block. The function is then used once again from code outside of the generate block. I'm not entirely sure what the spec. would require here, but my assumption is that a function declared within a generate block scope, would only be visible from within that scope, not outside of it. I ran Verible over the code and it was happy with the result, but Yosys complains that it "cannot resolve" the function. Please see attached:

[fail2.tar.gz](https://github.com/zachjs/sv2v/files/4962226/fail2.tar.gz)

zachjs, 2020-07-22:
These cast functions are permitted within generate blocks. I tried running your example, but it looks like `libv_macros.vh` is missing. With the troublesome macros commented out, the only usage of the cast function I can see is within the generate block itself.

stephenry, 2020-07-23:
I may be wrong (usually I am), but I suspect that the definition of `sv2v_cast_D66E2_signed` falls within a begin/end scope between lines 34 and 53 (corresponding to the if-statement). The 'generate' block itself does not terminate until line 83. I run the code through Verilator and it cannot see the definition for the cast function at line 58 which is outside of the original scope where it was declared. I suspect that SV2V needs to declare this cast before the if-statement so that it can be reused outside of it. 

Yosys on the other hand complains much earlier at line 47. I suspect that Yosys is wrong here, because if I hoist the cast out of the block it works. Looks like Yosys fails when functions are declared in these blocks.

Attached are all the files required to reproduce.

[fail3.tar.gz](https://github.com/zachjs/sv2v/files/4963264/fail3.tar.gz)

zachjs, 2020-07-23:
You are absolutely right that the cast within the generate conditional should not be visible outside that conditional. Prior to the insertion of that cast function, the "then" block of that conditional was comprised of a single item. sv2v was not correctly treating such single-item conditionals as an anonymous scope. This issue has been fixed as of 8537a9efdac7282c4d76993a4b6b8ab1a164f684.

I noticed, however, that your source contains items such as `wr_ptr_w` which are used before they are declared. While some tools (including Yosys) allow this, it is illegal per the Verilog specification. sv2v's conversion for `logic` turns certain items into `wire`s because their declarations occur after their usage in a procedural block.

stephenry, 2020-07-23:
That did the trick, thank you! 6.10 of IEEE1800 agrees with you. I'll need to go back and modify the files. Verible didn't appear to pick this up.

zachjs, 2020-08-08:
Are there outstanding requests related to this issue, @stephenry?

stephenry, 2020-08-09:
Not on my end.

closed 2020-08-09 by zachjs