zachjs/sv2v #117: type(foo) generated in output with unknown variables

rswarbrick, 2020-08-26:
This code is valid SystemVerilog (I think!):
```systemverilog
module a();
  always_comb begin
    if (foo == '0) foo = 'b1;
  end
endmodule
```
It has a stray reference to `foo`, which should be resolved based on whatever `default_nettype` is at the moment.

Since commit `9520894` ("unbased unsized conversion refactor"), this example gets transformed to:
```verilog
module a;
        always @(*)
                if (foo == {$bits(type(foo)) {1'sb0}})
                        foo = 'b1;
endmodule
```
which contains `type(foo)`, which isn't valid Verilog.

I think the problem is that the code in `UnbasedUnsized.hs` expands the `'0` to something that will work perfectly, so long as `foo` actually has a type. If I understand correctly, this would normally be resolved in `lookupTypeOf` in the `TypeOf` module but, since there is no definition of `foo`, we land in the `Nothing` case.

I did wonder about changing that function, but the problem is that we don't track `default_nettype`, so I don't think we can use the `Net` type: we don't know the right `NetType` value.

The solutions I could think of are either to add an "implicit" case to `NetType` or to add some sort of later pass that resolves `$bits(type(foo))` to `1` when `foo` is undefined. The latter option has the advantage that we don't need a net type at all, but the disadvantage that we'd have to add (yet) another conversion pass.

@zachjs: If this makes sense, but you don't have time to work on it at the moment, I'd happily do some of the work, but it would be good to get an idea of which approach you think is better. (Or if there's another more sensible approach)

@towoe, who spotted the problem in the first place and produced a small example showing it.

zachjs, 2020-08-27:
Firstly, I believe the given example is illegal, as `foo` is implicitly a `wire`, which cannot be procedurally assigned. That said, I recognize the code could be made legal while still relying on the implicit wire behavior.

How often do designs rely on `default_nettype` not being `none`? Though I'm not a hardware designer myself, I have often heard relying on implicit declarations is considered harmful. Further, how often are such nets used in combination with an unbased, unsized literal, where something like `'b0` would suffice?

I think my preferred solution would involve extending the `TypeOf` conversion to use the preceding or implied `default_nettype` directive. However, these resolutions could only proceed provided there are no remaining steps for the NestPI, Enum, or Package conversions to complete.

I'd be happy to work on this, though if it's urgent, my recommendation would be to employ a workaround in the SystemVerilog source.

towoe, 2020-08-27:
I think there are other cases in which it is not just an implicit declared wire where this goes wrong.
The original case we saw was a `` `define FOO !foo`` where the type of `` `FOO`` is not recognized.
Not sure in which way this differs for the internals.

The previous versions created a `'sd0` for the comparison, could this be a valid default?

From my side it is not urgent. Thanks @zachjs for working on this!

zachjs, 2020-09-14:
I've pushed a conversion which creates explicit declarations for implicit nets, which in turn allows these type expressions to be resolved. As I mentioned previously, the above example is not legal. However, something like
```SystemVerilog
module top;
    assign bar = 0;
    assign foo = bar == '1;
endmodule
```
now becomes
```SystemVerilog
module top;
	wire bar;
	assign bar = 0;
	wire foo;
	assign foo = bar == 1'sb1;
endmodule
```

> The original case we saw was a `define FOO !foo where the type of `FOO is not recognized.

I have also added support for type resolution of the logical negation operator. Thank you for pointing that out!

> The previous versions created a 'sd0 for the comparison, could this be a valid default?

While `'0` implicitly extends to be arbitrarily long, `'sd0` is actually 32 bits long. This might be a somewhat reasonable default, but I'd prefer fixing these type resolution issues directly, rather than potentially hiding unknown issues.

Please let me know if these changes work for you!

towoe, 2020-09-15:
Thanks a lot for working on this @zachjs!

I can confirm that this is working for various cases.
I tested it in the original environment where I encountered the problem and with multiple examples.

One thing I noticed is that for invalid SystemVerilog the output will still have something like `{$bits(type(bar)) {1'sb1}}` (if the line `assign bar = 0;` is removed from the previous example). Maybe it would make sense here to abort the conversion instead of inserting this in the code? I understand that sv2v should not be responsible for correcting invalid code, or even detecting it, but inserting this also makes no sense to me.

@zachjs feel free to close this issue as the changes provided are working for legal code.

zachjs, 2020-09-23:
Closing this issue given it has been resolved.

> One thing I noticed is that for invalid SystemVerilog the output will still have something like {$bits(type(bar)) {1'sb1}} (if the line `assign bar = 0;` is removed from the previous example). Maybe [...]

Many of the conversion phases are written to be simple and localized. For example, the conversion of unbased unsized literals doesn't do significant type resolution on its own. Rather than determining the context-determined size of the literals, the conversion can simply use `$bits(foo)`, and allow the other conversions to determine the type and size of `foo`. In this way, the conversion in question doesn't "know" whether `foo` is present or not, and so could not reliably catch and error out in this scenario.

This said, I can see value in something like a "pedantic" mode, wherein sv2v would check for common leftover SystemVerilog constructs (like `$bits` and `type`), and error out if any remain once the available conversion procedures have completed.

closed 2020-09-23 by zachjs