zachjs/sv2v #120: some assign expression in sv changed to always @(*) expression

zjuwlh, 2020-11-23:
Hi,

for some assign some assign expression in sv , this tool changes it to always @(*) expression, for example:
in sv:
assign singlebit_result    = '0;
 
after using sv2v, changed to :

always @(*)
  singlebit_result    = {32{1'sb0}};

this will be not good for using it in simulation, in the simulation , the signal will always be "x", since no variable changed in always   
block. How to fix it?

zachjs, 2020-11-24:
Could you give more context, especially the declaration and other usages of `singlebit_result`, ideally in a complete reproduction example? It is difficult to isolate potential issues or explore possible changes without additional information.

zjuwlh, 2020-11-24:
Hi,

you can try to use sv2v to change the following system verilog file to verilog file, the result verilog file will include the issue as I described.

**https://github.com/lowRISC/ibex/blob/master/rtl/ibex_alu.sv**

zachjs, 2020-11-25:
Depending on the value of the `RV32B` parameter for that module, the `logic [31:0] singlebit_result;` should either be converted to a `wire` or a `reg`. This ambiguity occurs because it is set using a procedural assignment in one case and using a continuous assignment in the other. Identifying such scenarios without instantiating the module may be outside the scope of sv2v.

Currently sv2v sees the procedural assignment, picks `reg`, and converts the continuous assignment to a procedural assignment accordingly. However, that event control indeed never fires because the right hand side is a constant. As of 260a650, when converting such continuous assignments, sv2v will now use `initial` rather than `always @*` when the RHS is trivially a constant expression. This appears to work as expected both for synthesis and simulation.

Please let me know if this change works for you!

zjuwlh, 2020-11-25:
is it possible to use **assign** rather than **always @*** ? since the original is **assign**

zjuwlh, 2020-11-25:
not use initial in such case, because I am not if initial can be synthesis.

zachjs, 2020-11-26:
> is it possible to use **assign** rather than **always @*** ? since the original is **assign**

Given that sv2v chose to convert `singlebit_result` to a `reg`, the continuous assignment _must_ be converted to a procedural assignment or the output would be malformed. While it is theoretically possible for sv2v to make the type of `singlebit_result` depend on the value of that parameter, this could add a lot of complexity.

> not use initial in such case, because I am not if initial can be synthesis.

While I was also skeptical about the use of `initial` as an alternative here, I have found that it works as expected in both simulation (using iverilog and others) and in synthesis (using Yosys).

As a proof of concept, I created a file `initial.v`:
```Verilog
module top(out);
    output reg out;
    initial out = 0;
endmodule
```
and ran
```Bash
yosys -p "read_verilog initial.v" -p "synth" -p "write_json out.json"
```
The resulting `out.json` indeed shows the output bit set appropriately.

This [blog post](http://billauer.co.il/blog/2018/02/verilog-initial-xst-quartus-vivado/) also suggests such `initial` blocks successfully synthesize with a few different commercial tools, though I have not verified this myself.

zjuwlh, 2020-11-26:
Ok, I will try to use commercial tools to verify it. Thanks.

zachjs, 2020-12-08:
@zjuwlh Any luck with the use of `initial` in the conversion?

zjuwlh, 2020-12-16:
thanks, so far it is ok. But I still think it's better to use "assign".

zachjs, 2021-01-28:
I recently encountered a commercial tool which doesn't support `initial`, and had to make some changes to how these scenarios are converted. In short, `always @*` is once again used, but the value is indirected through a temporary wire which appears to cause the block to fire correctly.

For example, the following
```SystemVerilog
module Example(
    input wire clk,
    input wire inp,
    output reg [2:0] out
);
    assign out[0] = 1'b0;
    always @(posedge clk)
        out[1] = inp;
endmodule
```
now becomes
```Verilog
module Example (
	clk,
	inp,
	out
);
	input wire clk;
	input wire inp;
	output reg [2:0] out;
	wire [1:1] sv2v_tmp_D6B46;
	assign sv2v_tmp_D6B46 = 1'b0;
	always @(*) out[0] = sv2v_tmp_D6B46;
	always @(posedge clk) out[1] = inp;
endmodule
```

Note that the `assign out[0] = 1'b0;` cannot be left as-is because although it is valid SystemVerilog, it is _not_ valid Verilog! SystemVerilog allows for at most one continuous assignment to an individual `reg`, but Verilog forbids any continuous assignments to a `reg`, outside of the so-called "procedural continuous assignments" `assign`/`deassign` and `force`/`release`.

zachjs, 2021-03-04:
Thank you for filing this issue! I am closing it because I believe it has been resolved and the discussion seems to have concluded. If this is not the case, please feel free to reopen this issue or file another!

closed 2021-03-04 by zachjs