zachjs/sv2v #122: sv2v fails when parsing interface parameters

wsipak, 2021-01-21:
example file:
https://github.com/lowRISC/opentitan/blob/c208de0094baf36288395beb455fb83e7290c9b1/hw/ip/otbn/dv/tracer/rtl/otbn_trace_if.sv

How to reproduce (with current master):

    git clone https://github.com/lowRISC/opentitan.git 
    cd opentitan
    git checkout c208de0094baf36288395beb455fb83e7290c9b1
    sv2v hw/ip/otbn/dv/tracer/rtl/otbn_trace_if.sv

Result:

    hw/ip/otbn/dv/tracer/rtl/otbn_trace_if.sv:27:30: Parse error: unexpected token ',' (Sym_comma).

Thats the part which fails:

    interface otbn_trace_if
    #(
      parameter int ImemAddrWidth,  // ',' is unexpected token
      parameter int DmemAddrWidth,
      parameter otbn_pkg::regfile_e RegFile = otbn_pkg::RegFileFF
    )(

Version of sv2v (current master):

    sv2v v0.0.6-2-ge94c034

zachjs, 2021-01-21:
sv2v doesn't currently support parameters without default values, but I am interested in adding support! I will look into this soon.

zachjs, 2021-01-21:
This issue should now be resolved! Please take a look and let me know if you run into any further issues.

closed 2021-01-22 by wsipak